LX3JL/xlxd #7: Repeaters are sorted in the first column of Modules table in users.php

yngwiechou, 2016-07-15:
I believe that the "Modules" table in users.php is intended to list user's nodes belong to the module under its column.

For example: Assuming there are two Modules in use, Node 1 & 2 is connected to Module A, while Node 3 on Module B

Module A   Module B
-------------------------
Node 1       Node 3
Node 2

However, the current code will display all the users, in the first column like this.

Module A   Module B
-------------------------
Node 1       
Node 2
Node 3

I managed to fixed the issue with the following change, please take look to see if it is appropriate.
Thanks!

=================================================

$ diff -u users.php users.php.fixed
--- users.php	2016-07-15 20:18:25.000000000 +0800
+++ users.php.fixed	2016-07-15 22:42:15.000000000 +0800
@@ -70,26 +70,37 @@

 echo '</tr>';

+$ModuleUsers = array();
 for ($i=0;$i<count($Modules);$i++) {
-
-    $Users = $Reflector->GetCallSignsInModules($Modules[$i]);
-
-   $UserCheckedArray = array();
-
-   for ($j=0;$j<count($Users);$j++) {
-      if (in_array($Users[$j], $UserCheckedArray)) {
-         $CurrentPositions = array_keys($UserCheckedArray,$Users[$j]);
-         $Displayname = $Users[$j].'-'.$Reflector->GetSuffixOfRepeater($Users[$j], $Modules[$i], max($CurrentPositions)+1);
-      }
-      else {
-         $Displayname = $Users[$j].'-'.$Reflector->GetSuffixOfRepeater($Users[$j], $Modules[$i]);
-      }
-      echo '
-            <tr>
-               <td><a href="http://www.aprs.fi/'.$Displayname.'" class="pl" target="_blank">'.$Displayname.'</a> </td>
-            </tr>';
-      $UserCheckedArray[] = $Users[$j];
-   }
+	$Users = $Reflector->GetCallSignsInModules($Modules[$i]);
+	$ModuleUsers[$i] = $Users;
+}
+
+// Find the max node count per module
+$MaxNodeRows = 0;
+foreach ($ModuleUsers as $mUsers) {
+	if ($MaxNodeRows < count ($mUsers)) {
+		$MaxNodeRows = count ($mUsers);
+	}
+}
+
+for ($k=0;$k<$MaxNodeRows;$k++) {
+	echo '<tr>';
+	for ($i=0;$i<count($Modules);$i++) {
+		$UserCheckedArray = array();
+		if ($ModuleUsers[$i][$k]) {
+			if (in_array($ModuleUsers[$i][$k], $UserCheckedArray)) {
+				$CurrentPositions = array_keys($UserCheckedArray,$ModuleUsers[$i][$k]);
+				$Displayname = $ModuleUsers[$i][$k].'-'.$Reflector->GetSuffixOfRepeater($ModuleUsers[$i][$k], $Modules[$i], max($CurrentPositions)+1);
+			}
+			else {
+				$Displayname = $ModuleUsers[$i][$k].'-'.$Reflector->GetSuffixOfRepeater($ModuleUsers[$i][$k], $Modules[$i]);
+			}
+			echo '<td><a href="http://www.aprs.fi/'. $Displayname .'" class="pl" target="_blank">' . $Displayname . '</a></td>';
+			$UserCheckedArray[] = $ModuleUsers[$i][$k];
+		}
+	}
+	echo '</tr>';
}

 ?>

yngwiechou, 2016-07-16:
After checking dashboard (1) cods, here a better fix for the issue.

However, I don't know how to shift the table-striped by 1, hence I used non-striped class for the enclosed table in the module <td>

73,
Yngwie
Bx2AFC

```
$ diff users.php users_fix2.php
54c54
<       <table class="table table-striped table-hover">
---
>       <table class="table table-striped">
72a73,74
> echo '<tr>';
>
75a78,81
>     echo '
>    <td valign="top" style="border:0px;padding:0px;">
>
>       <table class="table table-hover">';
88c94
<             <tr>
---
>       		<tr>
92a99,102
>    echo '
>       </table>
>
>    </td>';
94a105,106
> echo '</tr>';
>
```

lamaral, 2016-07-16:
Hi yngwiechou, sorry for the delay in the answer. I was out of reach this week.

Do you have this dashboard in use in a XLX server? I'm part of the XLX724 team and we only have users on Module B, so I have no way to test those changes.

I have some work that need to be done by tomorrow. I can take a look carefully on Monday.

phl0, 2016-07-16:
Luiz,

I can do tests if you like. I have various repeaters that can be linked to all modules on my XLX. Also some clients linke MMDVM, UP4DAR, DVmega...

   vy73 de Florian DF2ET

lamaral, 2016-07-16:
Hi Florian,

That would be really helpful. 

TY, 73

lamaral, 2016-07-16:
Hi Florian and Yngwie,

I made some changes to the code that I believe would fix that problem. Please try the code in my repo https://github.com/lamaral/xlxd to see if it fixes that problem.

Thank you.

phl0, 2016-07-16:
Luiz,

the fix works fine. This is what the current state in LX3JL repo looks like:
[image]

And this is the result from the test with your branch:
[image]

In both cases DB0BS-B was linked to my XLX module B. So the code was buggy and your fix is good. I think you can put it into a PR for the LX3JL repo.

   vy73 de Florian DF2ET

yngwiechou, 2016-07-17:
Hi Luiz,

The fix is works fine.

But if you can take a look my proposed fix #2, the presentation on web will be a little bit better

Thanks a lot.

Regards,
Yngwie

yngwiechou, 2016-07-17:
> Hi Luiz,
>
> The fix is works fine.
> <Screen Shot 2016-07-17 at 10.05.31.png>
>
> But if you can take a look my proposed fix #2, the presentation on web will be a little bit better
>
> <Screen Shot 2016-07-17 at 09.57.47.png>
>
> Thanks a lot.
>
> Regards,
> Yngwie

Hi Luiz,

Here my fix.

https://github.com/yngwiechou/xlxd

Please help to take a look to see if it is better.

Regards,
Yngwie

lamaral, 2016-07-30:
I believe this issue can be closed, as a fix was provided in the pull request #9.

lx1iq, 2016-07-31:
Hi guy's, Jean-Luc will close the issue when he is back from holidays.
Good work !

phl0, 2016-07-31:
@yngwiechou could close this issue as well as he was the author of this issue ;-)

lx1iq, 2016-07-31:
Of course :-)

closed 2016-09-04 by yngwiechou