LX3JL/xlxd #11: Link to bare IP address on peers page not always pointing to dashboard URL

phl0, 2016-07-29:
The column "IP" on the peers page of the dashboard contains a hyper link to the IP address of the reflector in case the IP addresses (or parts of it) are not hidden. That is not pointing to the dashboard in every case. 
For XLX518 for example this points to a completely different web page (the one of our amateur radio club) which does not directly relate to the XLX reflector or its dashboard.
Instead it would be nice to be able to reach the dashboard of the connected peers. Therefore I made a little patch. See https://github.com/LX3JL/xlxd/pull/10

lx1iq, 2016-07-31:
Hi good approach, it seams to work pretty fine.
When Jean-Luc is back from holydays we accept the pull request an close the issue.

phl0, 2016-07-31:
Roger that. As aligned further discussion in https://github.com/LX3JL/xlxd/pull/10

closed 2016-07-31 by phl0