LX3JL/xlxd #13: DCS protocol does not send last frame

yo2loj, 2016-08-15:
On outgoing DCS streams, the last frame is not encoded as a last frame.
Cause: incorrect conditions in cdcsprotocol.cpp, method HandleQueue:

```
if ( packet->IsDvFrame() )
            {
                EncodeDvPacket(m_DvHeadersCache[iModId], (const CDvFramePacket &)*packet, m_iSeqCounters[iModId]++, &buffer);
            }
            else if ( packet->IsLastPacket() )
            {
                EncodeDvLastPacket(m_DvHeadersCache[iModId], (const CDvFramePacket &)*packet, m_iSeqCounters[iModId]++, &buffer);
            }

```
Last packet is a DvFrame, so the else condition will never be reached.

Corrected code:
```
if ( packet->IsDvFrame() )
            {
                if ( packet->IsLastPacket() )
                {
                    EncodeDvLastPacket(m_DvHeadersCache[iModId], (const CDvFramePacket &)*packet, m_iSeqCounters[iModId]++, &buffer);
                }
                else
                {
                    EncodeDvPacket(m_DvHeadersCache[iModId], (const CDvFramePacket &)*packet, m_iSeqCounters[iModId]++, &buffer);
                }
            }
```
Patch file attached.

[cdcsprotocol.diff.zip](https://github.com/LX3JL/xlxd/files/419060/cdcsprotocol.diff.zip)

LX3JL, 2016-08-16:
Hi Marius,

You are correct.
I will incorporate your suggestion in next release, coming today or tomorow.
Thank you for your help

73
JL

yo2loj, 2016-08-17:
Solved in 1.3.8

closed 2016-08-17 by yo2loj