LX3JL/xlxd #21: DASHBOARD: Last heard displays module the user is currently connected to ...

g0wfv, 2016-11-29:
... not the module the user was last heard on.

LX1IQ, 2016-11-29:
Hi Tony,

I need a bit more details. 

73, Luc

g0wfv, 2016-11-29:
Hi Luc,

When a user Tx, the module column in last heard is populated with the module correctly (as this is the module the user is currently connected to and Txing on, right?)

As soon as the user changes module, their last heard entry reflects the new module they connected to (even if they didn't Tx) and if the user disconnects completely, the module is blank.

I'm currently having a look into the PHP to see if it's a quick fix, but not being familiar with your code (I only became aware of its existence and installed it up this afternoon - I'm impressed!) I'm a bit slow to work my way through the different files!

Steps to reproduce ...

1. Connect to Module A
2. Tx
3. Connect to Module B (or disconnect)

g0wfv, 2016-11-29:
For example ....

After Tx (on my local install):
[image]

After module change (with ircddbgateway remote app, so no Tx):
[image]

LX1IQ, 2016-11-29:
Hi Tony,

Thank you for the clarification. Yes that is a known behavior by design and cannot be changed as we do not use a database on the XLX server side.
The module is live parsed out of the call signe and the node where the user comes from.
At the moment the node is unlinked or it changes the module, the field stays blank or wrong until the station is transmitting again and the module can be refreshed.

73, Luc

g0wfv, 2016-11-29:
OK, as long as you are aware.  I'll close the issue ...

One more thing you might be able to point me in the correct direction with - my uptime doesn't populate (can be seen in the pics above) what have I mis-configured or not given xlxd permission to access ....?

g0wfv, 2016-11-29:
Disregard - I see it looks at the PID file which I won't have as I am running directly from command line and not as a service!

closed 2016-11-29 by g0wfv

LX1IQ, 2016-11-29:
Fine :-)

73, Luc