LX3JL/xlxd #40: XLX 2.0

iz7jwr, 2017-08-23:
Hi Luc, I am xlx332 and xlx023 's sysop. 
I have compiled and installed the new version of xlx, but the "start service" command does not start. Xlx023 is installed on a datacenter and is currently down.
73 de IZ7JWR Enrico.

iz7jwr, 2017-08-23:
All right, after editing the arguments in /init.d/xlxd. Current version XLX023 2.0 !!

yngwiechou, 2017-08-23:
Have an issue with:

/var/log/xlxd.xml

Empty content will be generated every 15 seconds.

Regards,
Yngwie
BU2EA

yngwiechou, 2017-08-23:
It seems xlxd failed to GetPeers() and broke the WriteXmlFile()

xmlFile << "<" << m_Callsign << "linked peers>" << std::endl;

<----------- WriteXmlFile break here before GetPeers()
// lock
CPeers *peers = GetPeers();
// iterate on peers
for ( int i = 0; i < peers->GetSize(); i++ )
{
	peers->GetPeer(i)->WriteXml(xmlFile);
}

iz7jwr, 2017-08-23:
I can not connect MMDVM. With the parameters of my XLX023 into MMDVM.ini. The contents of the Log / var / log / messages file are:
"Aug 23 05:17:59 v12356 xlxd: DMRmmdvm connect packet from unrecognized id 2227145"
"Aug 23 05:17:59 v12356 xlxd: DMRmmdvm packet (8)"

yngwiechou, 2017-08-23:
Ok, the issue was caused by a flapping peer.

After removing the problematic interlink peer, GetPeers() works well.

But improvement in GetPeers() is still needed in handling the case with unstable interlinks.

Regards,
Yngwie
BU2EA

phl0, 2017-08-23:
You need to have a valid dmrids.dat file containing the DMR IDs and callsigns.

yngwiechou, 2017-08-23:
main.h:#define DMRIDDB_PATH                    "/xlxd/dmrid.dat”

phl0, 2017-08-23:
And #define DMRIDDB_USE_RLX_SERVER should be set to 1 or otherwise create that file manually.

yngwiechou, 2017-08-23:
Use the command below to generate the dmrid.dat

/xlxd# wget -O dmrid.dat http://xlxapi.rlx.lu/api/exportdmr.php

LX1IQ, 2017-08-23:
Hi guys,

I updated the install doc in the group a few minutes ago.
Please have a look.

73, Luc

DECHARGE

Ce message et toutes pièces jointes sont établis à l'intention exclusive de ses destinataires. Ils peuvent contenir des informations confidentielles. Si vous recevez ce message par erreur, merci de le détruire et d'en avertir immédiatement l'expéditeur. Toute utilisation de ce message non conforme à sa destination, toute diffusion ou toute publication, totale ou partielle est interdite, sauf autorisation expresse. Ce message a fait l'objet d'un traitement anti-virus.

Le contenu de ce message et des pièces jointes ne pourrait engager la responsabilité de l’AGEDOC asbl que s'il a été émis par une personne dûment habilitée agissant dans le strict cadre des fonctions auxquelles elle est employée et à des fins non étrangères à ses attributions.

LX1IQ, 2017-08-23:
Please read the installation doc in the yahoo group ! You need a valid dmrid.dat in the xlxd/ folder.
I will update the readme in the next days.

73, Luc

closed 2017-08-23 by LX1IQ