LX3JL/xlxd #55: Learning and Support Resources

mrickey, 2017-09-12:
Are there any resources to help me learn more about XLX 2.0? I want to learn the supported protocols and about the protocol conversion server (like DMR to analog or D-STAR).

Is there a yahoo or google group? What about a Facebook or Google+ page? A wiki, anything?

Currently we have three reflectors and it's getting to complex to manage them. I'm hoping we can do everything with one XLXD 2.0 reflector.

Thanks in advance,
Michael AF6FB

phl0, 2017-09-13:
The XLX does not do anything analog. It currently supports DMR and D-Star. There is a Yahoo group for XLX: https://groups.yahoo.com/neo/groups/xlxd-star/

mrickey, 2017-09-13:
> groups/xlxd-star/

I'm getting a 404 when I use that link.

I thought the XLX could connect to an allstar server, guess I was wrong.

*Michael Rickey* - AF6FB
949-298-0529 (*voice, sms, allo, duo*)
michael.rickey@gmail.com

phl0, 2017-09-13:
Watch out for a Yahoo group named xlxd-star

johnhays, 2017-09-13:
Yes, the Yahoo group exists.  It should be moved to groups.io which has
github integration.  (Yahoo! groups may disappear under new ownership.)