LX3JL/xlxd #64: Dont Start

ameyst, 2017-11-15:
308/5000
Hello,
I have done the installation with the readme
nothing appears in the process list and the xlxd.xml is not created.

root @ s265269: ~ # service xlxd start
Starting xlxd:
root @ s265269: ~ #

the server is a V-Server with Debian 7.11 64bit

I have no idea what's wrong

greetings stefan

phl0, 2017-11-24:
What is the output if you start the process by hand?

ameyst, 2017-11-24:
nothing and the pit is empty

phl0, 2017-11-24:
Screenshot please

ameyst, 2017-11-24:
root@s265269:~# cd /xlxd/               
root@s265269:/xlxd# ./xlxd --chuid root:root  -- XLX263 78.46.11.69
root@s265269:/xlxd# ./xlxd --chuid root:root --XLX263 78.46.11.69
root@s265269:/xlxd# /etc/init.d/xlxd start                       
Starting xlxd: 
root@s265269:/xlxd#

phl0, 2017-11-24:
You need to add the ip address of the transcoder. If it is not present try 127.0.0.1 instead.
Also the chuid stuff does not make sense to me. Just try "xlxd XLX263 78.46.11.69 127.0.0.1".

LX1IQ, 2017-12-19:
I guess we can close this issue.

73, Luc

closed 2017-12-19 by LX1IQ