LX3JL/xlxd #73: Request: pass original DMR IDs to XLX interlink

cyanide-burnout, 2018-01-15:
At this moment XLX interlink uses D-STAR headers only to pass traffic between reflector. If user has more than one ID a reflector system on the other side may use another ID than used on original call.

LX1IQ, 2018-01-15:
I know that there are users who have several ID, but can you explain why?

cyanide-burnout, 2018-01-15:
For example they have several radios of different vendor and so on. This is existing situation that cannot be ignored.

LX1IQ, 2018-01-15:
So you mean that those people are talking at the same time with two different radios and different ID's but the same callsinge ??? Ok, let's make it complicated ??

cyanide-burnout, 2018-01-15:
I mean users may have more that single ID in general. In this case a xlxd gets the QSO with one source ID but after this call will be transferred to the second xlxd, the last one can set second (or third) ID to the outbound DMR transmission.
In general it could be OK but... Since xlxd passes embedded signalling AS-IS the call receiver may be in trouble due to values of source ID of voice header and embedded signalling may differ.
So it would be good if xlxd will enrich DSVT voice header packet with Full LC on the same way as it does with DV Frame packets.

LX1IQ, 2018-01-15:
Ok, we are going to see what we can do.
73, Luc

vk4tux, 2019-06-02:
Would Artems request also mean NXDN id numbers from BM be passed via xlxd.interlink to XLX where 5 digit id has been added to dmrid.dat (added manually) & be recognised for correct XLX DB call display     
?

vk4tux, 2019-07-22:
I resolved my issues with BM xlxd.interlink, by removing it replacing it with BM openbridge and hblink peers. Now all traffic enters xlxd, and all the qso is heard.

citynetcomputing, 2021-01-27:
Just for more clarity on the having multiple ID's. It is a requirement to have multiple ID's if you own different vendors of radio. I own Motorola and Chinese radios and use full APRS/GPS and TMS messaging on both. To set them up properly in Brandmeister for routing it requires separate DMR ID's in the Brandmeister system.