LX3JL/xlxd #74: Request: pass connection option on XLX interlink to force DMR to D-STAR transcoding

cyanide-burnout, 2018-01-15:
Existing version of XLX Interlink has requirement to pass D-STAR and DMR traffic at the same time.
Interlink can be forced to do D-STAR to DMR transcoding by downgrading version number.
Please add capability to force DMR to D-STAR conversion.

cyanide-burnout, 2018-01-17:
My existing idea is to reuse revision number to pass something like PROTOCOL_VERSION_2_SPECIAL to force DMR to D-STAR transcoding. For example I can use revisions > 200 to force that.
If you OK with this approach I can prepare pull request.

stefansaraev, 2018-01-23:
@cyanide-burnout @LX3JL c3920519872deb15cf7afc3673221d0541085cdf works for me. thanks guys.

LX1IQ, 2018-01-23:
Hi,
Tnx for the feedback. I guess Artem can close this one.

73,  Luc

closed 2018-01-23 by cyanide-burnout