LX3JL/xlxd #75: Future request / issue report: care about packet reordering

cyanide-burnout, 2018-01-15:
Existing implementations of CPacketStream and CPacketQueue don't care about probabilistic reordering of UDP packetsthey do not care about probabilistic reordering of UDP packets.
Normally it should care about sequence numbers to process audio frames on the right order.

LX3JL, 2019-03-16:
This is actually the job of the repeaters/hotspot.
Indeed, even if xlx reorder udp packet before forwaring them to clients, there are no
garanties that they will not be scrambled again on their way to the final destination.

cyanide-burnout, 2019-03-17:
I cannot be agree with your point. In case when xlxd does transcoding it changes sequence numbers from D-STAR’s resolution of 21 numbers per sequence to at least MMDVM’s resolution of 256 numbers per sequence. In this case you breake a sequence. Also AMBE has a statefull algorithm and you have to provide frames in proper sequence.

cyanide-burnout, 2019-03-17:
So I meant that not only repeaters/hotspots should care about sequence but xlxd too