LX3JL/xlxd #77: Feature reqeust: Define the Module which has to be transcoded

oe5stm, 2018-01-21:
Hi Luc,

i think that it is important for the future to have the possibility to choose the Module which is been transcoded.
Every stream right now is send to the AMBE-Server and so i get stuck with only 4 Channels available.

Bye
Timm

ik5xmk, 2018-01-22:
The strength of a reflector like XLX is the possibility of having separate modules. But it is important to think in the future (not too remote, hope) to a flexible management of the same. Then define (as I authorize a peer) on which module transcoding is present and which protocols are active. As the interlink file is present, there should be a file in which to manage the configuration of the modules. Thanks, 73. David IK5XMK

iw2gob, 2018-01-22:
Hi all,  
I believe that the possibility of choosing the module that needs to be transcoded is fundamental and can even heavily influence the future of the XLX multi-protocol system. Some XLX networks in the world have different needs and the flexibility of the configuration is a very important aspect that should not be underestimated.
I hope Luc can soon change his point of view on this issue.

73, Maurizio IW2GOB

oe1kbc, 2018-01-27:
Hi,
i found a quick solution:

change source cpacketstream.cpp

bool CPacketStream::Open(const CDvHeaderPacket &DvHeader, CClient *client)
{
    bool ok = false;
   
    // not already open?
    if ( !m_bOpen )
    {
        // update status
        m_bOpen = true;
        m_uiStreamId = DvHeader.GetStreamId();
        m_uiPacketCntr = 0;
        m_DvHeader = DvHeader;
        m_OwnerClient = client;
        m_LastPacketTime.Now();
		if(DvHeader.GetRpt2Module() == 'A' || DvHeader.GetRpt2Module() == 'M')
			m_CodecStream = g_Transcoder.GetStream(this, client->GetCodec());
		else
			m_CodecStream = g_Transcoder.GetStream(this, CODEC_NONE);
        ok = true;
    }
    return ok;
}

73 de Kurt
OE1KBC

IZ7AUH, 2018-01-28:
Kurt,
many tnx I put your code into my XLX099 on module U and now a Trasconde is locked yo U only...

stefansaraev, 2018-01-28:
there is a quick working solution in my git.

vk4tux, 2018-01-29:
Thankyou Stefan You have some good code there ;
root@XLX389:~# git clone https://github.com/stefansaraev/xlxd.git xlxd_stefan
Cloning into 'xlxd_stefan'...

ik5xmk, 2018-01-29:
Thank you very much to all, hoping that Luc can implement these changes with activation, perhaps, from the command line or similar, in the next version. 73, David

citynetcomputing, 2019-12-30:
Ditto!

closed 2020-01-04 by LX1IQ