LX3JL/xlxd #83: system paths like WL/BL/XML are hardcoded

shawnchain, 2018-02-26:
I would like to install xlxd under /opt/xlxd but no luck.

Would appreciate if the system path are configurable via startup parameter, or introduce some config file would be great.

phl0, 2020-01-23:
You can change the settings in main.h to choose which folders the binary shall look for BL/WL. 
And you may also adapt the Makefile so that "make install" installs to /opt/xlxd/". If you need a template, let me know.

fabianfrz, 2024-06-09:
@shawnchain I wrote an ansible playbook which patches the main.h and the makefile. You can copy the executable to any directory you like but before you compile, you need to adjust your settings in main.h. I currently do that via ansible.

The idea to have some config file, that is read during startup would be better than compile the settings into the binary. It is easier to change for the sysop.