LX3JL/xlxd #86: Can't open several sockets

Desertmansd, 2018-03-15:
There are several sockets that xlxd can't open. The firewall has been set properly and no other programs are using those sockets. 

Thank you
73
Ted

[image]

phl0, 2018-03-16:
What do "netstat -ulnvp" and "ip -4 addr show" tell?
Maybe you are using the official IPv4 address whereas the system is NATted behind? Then you have to use the private IP.

phl0, 2018-03-16:
Sure. How else would the process bind an IP address that is not assigned to one of its interfaces? ;-)

Desertmansd, 2018-03-16:
Very true. I don't it was clear in the README that xlxd uses that config file to read it's own ip address as I've seen many a program ask for the routable ip in a config so it can report back to a server.