LX3JL/xlxd #88: Multiple Devices

johnhays, 2018-05-13:
While testing the new build of the LX3JL USB-3006 device, under ambed 1.3.1 observed two conditions:
1. If I change the product description string from USB-3006 to XC-3006 and modify the cftdidevicedescr.cpp to recognize that string, on startup it gets an error:

**Opening USB-3006_A:(serial number) device
FTDI function FT_OpenEx error 2**

2. If additionally 2 ThumbDVs are plugged in along with the USB-3006, it sees both the XC-3006 and ThumbDV devices, but only reads the USB-3006 and never completes initializing the server. (Is this a feature or bug?)  It works fine with the 2 ThumbDVs or USB-3006 independently.

The good news is the sample production unit is performing well with the software.  We are still doing some hardware testing in the lab, but if no issues are found, the production run should ship right after Hamvention (watch the NW Digital Radio blog at http://nwdigitalradio.com )

LX3JL, 2018-05-14:
Hi John,

Does it works when the device ID is set to USB-3006?

73
JL

johnhays, 2018-05-14:
Yes it works with USB-3006 as product description.

It doesn't work with a mix of USB-3006 and ThumbDVs.

LX3JL, 2018-05-14:
USB-3006 vs XC-3006:
==================

So, I guess that the issue of not working with XC-3006 device description string
is within the ambed software. Keep in mind that 3006 devices are multi-interface
as they have two 3003 chip per device. So code source mods involve a bit more than adding a 3003
device. The good approach is probably to code a new “Instantiate” function.
Do you really want to stick to XC-3006 ? What is the added value compared to existing USB-3006 ?
I can help with modifying the code if you need it, but not before the week-end.

USB-3006 coexisting with ThumbDV:
============================

This works. I tested it on a long run (XLX950 did run on one USB-3006 + 2 ThumbDV for a few weeks).
However, I found the setup only working when plugging the devices directly on the PC (ie no USB hubs)
Generally speaking, ThumbDV (and other 3003 clones) very sensitive to hubs.

73
JL

johnhays, 2018-05-14:
We can live with the USB-3006 string, though I think with some refactoring
any string should be supportable. It might even be wise to have a hash
table to associate different strings with instantiation functions.

I will test mixed devices without a hub.

johnhays, 2018-05-14:
Starting AMBEd 1.2.1

Initializing vocodecs:
Detected 4 USB-FTDI devices

Description : USB-3006 A	 Serial : XC2AJAJTA
Description : USB-3006 B	 Serial : XC2AJAJTB
Description : ThumbDV	 Serial : DN034P21
Description : ThumbDV	 Serial : DN034P27
Opening USB-3006_A:XC2AJAJTA device
ReadDeviceVersion : 0AMBE3003F1V120.E100.XXXX.C106.G514.R008.A0030608.C0020208/

Opening USB-3006_B:XC2AJAJTA device
ReadDeviceVersion : 0AMBE3003F1V120.E100.XXXX.C106.G514.R008.A0030608.C0020208/

Opening ThumbDV:DN034P21 device
ReadDeviceVersion : 0AMBE3000R1V120.E100.XXXX.C106.G514.R009.B0010411.C0020208

Opening ThumbDV:DN034P27 device
ReadDeviceVersion : 0AMBE3000R1V120.E100.XXXX.C106.G514.R009.B0010411.C0020208

Codec interfaces initialized successfully : 8 channels availables

Initializing controller

AMBEd started and listening on 127.0.0.1
Stream Open from XLX000  
Vocodec channel USB-3006_A:1 -> USB-3006_A:0 open
Opened stream 1
Stream Open from XLX000  
Vocodec channel USB-3006_B:1 -> USB-3006_B:0 open
Opened stream 2
Stream Open from XLX000  
Vocodec channel USB-3006_B:2 -> USB-3006_A:2 open
Opened stream 3
Stream Open from XLX000  
Vocodec channel ThumbDV:0 -> ThumbDV:0 open
Opened stream 4
Stream Open from XLX000  
Vocodec channel USB-3006_A:0 -> USB-3006_A:1 open
Opened stream 5
Stream Open from XLX000  
Vocodec channel USB-3006_B:0 -> USB-3006_B:1 open
Opened stream 6
Stream Open from XLX000  
Vocodec channel USB-3006_A:2 -> USB-3006_B:2 open
Opened stream 7
Stream Open from XLX000  
Vocodec channel ThumbDV:0 -> ThumbDV:0 open
Opened stream 8
Stream 1 activity timeout 
Vocodec channel USB-3006_A:1 -> USB-3006_A:0 closed
0 of 1683 packets lost
Stream 2 activity timeout 
Vocodec channel USB-3006_B:1 -> USB-3006_B:0 closed
0 of 1668 packets lost
Stream 3 activity timeout 
Vocodec channel USB-3006_B:2 -> USB-3006_A:2 closed
0 of 1653 packets lost
Stream 4 activity timeout 
Vocodec channel ThumbDV:0 -> ThumbDV:0 closed
0 of 1638 packets lost
Stream 5 activity timeout 
Vocodec channel USB-3006_A:0 -> USB-3006_A:1 closed
0 of 1623 packets lost
Stream 6 activity timeout 
Vocodec channel USB-3006_B:0 -> USB-3006_B:1 closed
0 of 1608 packets lost
Stream 7 activity timeout 
Vocodec channel USB-3006_A:2 -> USB-3006_B:2 closed
0 of 1593 packets lost
Stream 8 activity timeout 
Vocodec channel ThumbDV:0 -> ThumbDV:0 closed
0 of 1578 packets lost

johnhays, 2018-05-14:
I recompiled to move up from 1.2.1 and got

Starting AMBEd

Initializing vocodecs:
Detected 4 USB-FTDI devices

Description : USB-3006 A	 Serial : XC2AJAJTA
Description : USB-3006 B	 Serial : XC2AJAJTB
Description : ThumbDV	 Serial : DN034P21
Description : ThumbDV	 Serial : DN034P27
Opening USB-3000:DN034P21 device
ReadDeviceVersion : 0AMBE3000R1V120.E100.XXXX.C106.G514.R009.B0010411.C0020208

Opening USB-3000:DN034P27 device
ReadDeviceVersion : 0AMBE3000R1V120.E100.XXXX.C106.G514.R009.B0010411.C0020208

Codec interfaces initialized successfully : 2 channels availables

Initializing controller

AMBEd started and listening on 127.0.0.1

johnhays, 2018-05-14:
I will completely rebase the source to 1.3.1 and test later today.

johnhays, 2018-05-14:
OK -- looks good with 1.3.1 using USB-3006 and a couple of ThumbDVs (no hub).

Starting AMBEd 1.3.1

Initializing vocodecs:
Detected 4 USB-FTDI devices

Description : USB-3006 A	 Serial : XC2AJAJTA
Description : USB-3006 B	 Serial : XC2AJAJTB
Description : ThumbDV	 Serial : DN034P21
Description : ThumbDV	 Serial : DN034P27
Opening USB-3006_A:XC2AJAJTA device
ReadDeviceVersion : 0AMBE3003F1V120.E100.XXXX.C106.G514.R008.A0030608.C0020208/

Opening USB-3006_B:XC2AJAJTA device
ReadDeviceVersion : 0AMBE3003F1V120.E100.XXXX.C106.G514.R008.A0030608.C0020208/

Opening ThumbDV:DN034P21 device
ReadDeviceVersion : 0AMBE3000R1V120.E100.XXXX.C106.G514.R009.B0010411.C0020208

Opening ThumbDV:DN034P27 device
ReadDeviceVersion : 0AMBE3000R1V120.E100.XXXX.C106.G514.R009.B0010411.C0020208

Codec interfaces initialized successfully : 8 channels availables

Initializing controller

AMBEd started and listening on 127.0.0.1
Stream Open from XLX000  
Vocodec channel USB-3006_A:1 -> USB-3006_A:0 open
Opened stream 1
Stream Open from XLX000  
Vocodec channel USB-3006_B:1 -> USB-3006_B:0 open
Opened stream 2
Stream Open from XLX000  
Vocodec channel USB-3006_B:2 -> USB-3006_A:2 open
Opened stream 3
Stream Open from XLX000  
Vocodec channel ThumbDV:0 -> ThumbDV:0 open
Opened stream 4
Stream Open from XLX000  
Vocodec channel USB-3006_A:0 -> USB-3006_A:1 open
Opened stream 5
Stream Open from XLX000  
Vocodec channel USB-3006_B:0 -> USB-3006_B:1 open
Opened stream 6
Stream Open from XLX000  
Vocodec channel USB-3006_A:2 -> USB-3006_B:2 open
Opened stream 7
Stream Open from XLX000  
Vocodec channel ThumbDV:0 -> ThumbDV:0 open
Opened stream 8
Stream 1 activity timeout 
Vocodec channel USB-3006_A:1 -> USB-3006_A:0 closed
0 of 1354 packets lost
Stream 2 activity timeout 
Vocodec channel USB-3006_B:1 -> USB-3006_B:0 closed
0 of 1339 packets lost
Stream 3 activity timeout 
Vocodec channel USB-3006_B:2 -> USB-3006_A:2 closed
0 of 1324 packets lost
Stream 4 activity timeout 
Vocodec channel ThumbDV:0 -> ThumbDV:0 closed
0 of 1309 packets lost
Stream 5 activity timeout 
Vocodec channel USB-3006_A:0 -> USB-3006_A:1 closed
0 of 1294 packets lost
Stream 6 activity timeout 
Vocodec channel USB-3006_B:0 -> USB-3006_B:1 closed
0 of 1279 packets lost
Stream 7 activity timeout 
Vocodec channel USB-3006_A:2 -> USB-3006_B:2 closed
0 of 1264 packets lost
Stream 8 activity timeout 
Vocodec channel ThumbDV:0 -> ThumbDV:0 closed
0 of 1249 packets lost

closed 2018-05-14 by johnhays