LX3JL/xlxd #94: xlxd segfaults?

sm0tsc, 2018-06-26:
I have seen this

Merlinv2, 2019-11-24:
I found this same problem on some server ..
nobody knows how to fix this .. it's strange

vk4tux, 2019-11-24:
Does this change resolve the issue ?;'

src/cprotocol.cpp 
<https://github.com/mreckhof/xlxd/commit/0a6a423522d1249078edf769c2ddebcf742931f7#diff-33db1e5e0c70dc811df8c1542d5a5a69>
<https://github.com/mreckhof/xlxd/commit/0a6a423522d1249078edf769c2ddebcf742931f7#diff-33db1e5e0c70dc811df8c1542d5a5a69>
	@@ -176,11 +176,13 @@ CPacketStream *CProtocol::GetStream(uint16 
uiStreamId, const CIp *Ip)

	
	{

	
	if ( m_Streams[i]->GetStreamId() == uiStreamId )

	
	{

	
	if (m_Streams[i]->GetOwnerIp()) {

	
	// if Ip not NULL, also check if IP match

	
	if ( (Ip != NULL) && (*Ip == *(m_Streams[i]->GetOwnerIp())) )

	
	{

	
	stream = m_Streams[i];

	
	}

	
	}

	
	}

	
	}

	
	// done

Merlinv2, 2019-11-24:
I am going to try this.. thx!!

Merlinv2, 2019-11-25:
> 
> Does this change resolve the issue ?;' src/cprotocol.cpp <[mreckhof@0a6a423#diff-33db1e5e0c70dc811df8c1542d5a5a69](https://github.com/mreckhof/xlxd/commit/0a6a423522d1249078edf769c2ddebcf742931f7#diff-33db1e5e0c70dc811df8c1542d5a5a69)> <[mreckhof@0a6a423#diff-33db1e5e0c70dc811df8c1542d5a5a69](https://github.com/mreckhof/xlxd/commit/0a6a423522d1249078edf769c2ddebcf742931f7#diff-33db1e5e0c70dc811df8c1542d5a5a69)> @@ -176,11 +176,13 @@ CPacketStream *CProtocol::GetStream(uint16 uiStreamId, const CIp *Ip) { if ( m_Streams[i]->GetStreamId() == uiStreamId ) { if (m_Streams[i]->GetOwnerIp()) { // if Ip not NULL, also check if IP match if ( (Ip != NULL) && (*Ip == *(m_Streams[i]->GetOwnerIp())) ) { stream = m_Streams[i]; } } } } // done
> […](#)

compiling with the changes made gives error...

LX1IQ, 2019-12-01:
Hi folks,
XLX 2.3.3 is only and should address this issue.
Hope it helps.
Any feedback is highly appreciated.
73, Luc

Merlinv2, 2019-12-02:
> 
> 
> Hi folks,
> XLX 2.3.3 is online and should address this issue.
> Hope it helps.
> Any feedback is highly appreciated.
> 73, Luc

last afternoon I did update .. for now 16 hours online..thanks!

closed 2020-01-04 by LX1IQ