LX3JL/xlxd #95: How to distinguish a DMR vs DStar key up

sixfootseven, 2018-07-01:
I am trying to have a live display on our dashboard of whether the current talker is using DMR or DStar, and I can't find a reliable method. I can't use the callsign of their repeater or gateway, as some users have the same callsign and band for 2 different connections - DMR and DStar. Can the syslog entries please have the mode somewhere in the entry? Or is there another way to tell? 

Jul  1 20:58:15 xlx xlxd: Opening stream on module A for client ZL2EZ  B with sid 3150

vk4tux, 2018-07-01:
> Reply-to: LX3JL/xlxd <reply
> +00854660b27426a4b4934785fc3529c5b1146004523158d692cf00000001175061a292a169ce141a724f@reply.github.com>

The suffix box indicates a suffix entry for the dstar user and no suffix
for the dmr user.

sixfootseven, 2018-07-01:
Cheers, figured i might have missed something obvious. Thanks Adrian.

closed 2018-07-01 by sixfootseven

vk4tux, 2018-07-01:
If 'DMR' is seen in the suffix field, it indicates a BM TG source via
XLX Interlink.

+00854660ebda7577a20c1deb862cbb83a85c2e7ab741247192cf00000001175077dd92a169ce141a724f@reply.github.com>
Cc: Adrian Fewster <vk4tux@gmail.com>, Comment
<comment@noreply.github.com>
Date: Sun, 01 Jul 2018 04:07:09 -0700