LX3JL/xlxd #103: Losing audio requiring restart

sixfootseven, 2018-07-30:
At random, our xlxd 2.2.2 sometimes stops passing audio, and requires a service restart to solve. It might go 15 days without this behaviour, or it might be 5 hours. No clues in the syslog, everything looks normal.

Ubuntu 18.04 LTS
Intel Atom d525mw Dual Core
4GB RAM (3.5GB free)

I have git cloned, and re-compiled from scratch, and still having the same problem.

sm0tsc, 2018-08-07:
> I suspect the package fail2ban could have something to do with it. I
> installed it approximately the same time this issue started. I have
> uninstalled fail2ban and will see how it goes.

If fail2ban is responsible you should been able to see the blocks in
iptables

//SM0TSC - Johan

sixfootseven, 2018-08-21:
I have not re-activated fail2ban to wait for the problem to occur so i can check iptables, but at around 16 days uptime now, it looks like the problem has disappeared, with the deactivation of fail2ban the only change in that time. Closing.

closed 2018-08-21 by sixfootseven