LX3JL/xlxd #108: After upgrading to #107 Dashboard does not refresh automatically

sm0tsc, 2018-10-24:
$PageOptions['PageRefreshActive']                    = true;   			// Activate automatic refresh

Is set to true

//Johan - SM0TSC

LX1IQ, 2018-10-24:
Hi Johan,
Are you sure? I tested with edge and firefox without problems.

73, Luc

sm0tsc, 2018-10-24:
I am using Chrome.. Will try edge...
//Johan - SM0TSC

LX1IQ, 2018-10-24:
Just works fine at me with chrome too.
Hmm, can you check https://xlx270.epf.lu  ?

sm0tsc, 2018-10-24:
It does not update in any browers..

I will keep monitor your XLX

Can you check http://xlx.brandmeister.se

//Johan - SM0TSC

LX1IQ, 2018-10-24:
I see.
Try to put some text into the config.inc.php 
$PageOptions['CustomTXT']                            = 'test';					// custom text in your header

sm0tsc, 2018-10-24:
Done
//Johan - SM0TSC

LX1IQ, 2018-10-24:
your jquery is not loading correctly !
replace the jquery-1.12.4.min.js in your "js" folder without editing it.
________________________________________________________________________________________

HTML1300: Navigation occurred.    xlx.brandmeister.se (1,1)
HTML1420: Unexpected quote after "PUBLIC" or "SYSTEM" keyword. Expected whitespace.
xlx.brandmeister.se (1,22)

SCRIPT1002: SCRIPT1002: Syntax error   jquery-1.12.4.min.js (7,1)
0: '$' is not defined  xlx.brandmeister.se (20,9)

sm0tsc, 2018-10-24:
Ok done...

//Johan - SM0TSC

LX1IQ, 2018-10-24:
seems to be ok for now.

sm0tsc, 2018-10-24:
Now seems to update.. Also Server uptimes reload correctly.. Thank you...

LX1IQ, 2018-10-24:
you're welcome and please close the issue.

closed 2018-10-24 by sm0tsc