LX3JL/xlxd #115: Crash streams when the communication between xlxd and ambed is unstable

lucamarche-iz1mlt, 2018-11-30:
Hi everyone,

I have an xlx with ambed software on a second machine, when the connection between xlxd and ambed starts to have a very variable latency the streaming ambe xlx side crashes and no longer resumes, the only way is to restart xlxd.

By increasing the TRANSCODER_AMBEPACKET_TIMEOUT and STREAM_TIMEOUT the thing gets better but the fact is that when it crashes it does not recover.

Adding some prints in the file ctranscoder.cpp, forcing the printing of ambed stats and packet timeouts,  noticed that the ping goes to -1000.0 and changes the order of the printouts, but the keepalive communication remains sent, the operation of which remains unchanged. If anyone wants to take a look I attach log files.

Could it be some problem of managing any counter or lack of management of a state?

And some time beating myself but have not come to terms with it yet.

IZ1MLT Luca
[log xlxd.txt](https://github.com/LX3JL/xlxd/files/2632854/log.xlxd.txt)

ea5gvk, 2018-12-01:
Hi everyone,
The same problem happens to me.
Some help to solve this problem in my XLX051.
´73 EA5GVK.

lucamarche-iz1mlt, 2018-12-04:
After several attempts to put the code in order to find a solution to the problem, it seems that by adjusting the various time out the phenomenon is resolved.

From various tests done in the presence of latency with a variation of 2 seconds the communication stops and when the next stream reopens, it starts again correctly.

Here I report the changes made. Someone try them and check that they behave the same way.

-------------xlxd---------------
in main.h
#define TRANSCODER_KEEPALIVE_PERIOD     1                                   // in seconds
#define TRANSCODER_KEEPALIVE_TIMEOUT    (TRANSCODER_KEEPALIVE_PERIOD*3)     // in seconds
#define TRANSCODER_AMBEPACKET_TIMEOUT   (TRANSCODER_KEEPALIVE_TIMEOUT*1000)  // in ms

in ctranscoder.cpp
#define AMBED_OPENSTREAM_TIMEOUT    TRANSCODER_AMBEPACKET_TIMEOUT   // in ms

in cpacketstream.h
#define STREAM_TIMEOUT      (1.000*TRANSCODER_AMBEPACKET_TIMEOUT)

OR

in main.h
#define TRANSCODER_KEEPALIVE_PERIOD     2                                   // in seconds
#define TRANSCODER_KEEPALIVE_TIMEOUT    (TRANSCODER_KEEPALIVE_PERIOD*2)     // in seconds
#define TRANSCODER_AMBEPACKET_TIMEOUT   (TRANSCODER_KEEPALIVE_PERIOD *1000)  // in ms

in ctranscoder.cpp
#define AMBED_OPENSTREAM_TIMEOUT    TRANSCODER_AMBEPACKET_TIMEOUT   // in ms

in cpacketstream.h
#define STREAM_TIMEOUT      (1.000*TRANSCODER_AMBEPACKET_TIMEOUT)
---------------------------------------------------------------
in ccodecstream.cpp

    // any packet from transcoder
    if ( m_Socket.Receive(&Buffer, &Ip, 20) != -1 )  change from 5 to 20,

--------------ambed----------

in main.h
#define TRANSCODER_KEEPALIVE_PERIOD     1                                   // in seconds
#define TRANSCODER_KEEPALIVE_TIMEOUT    (TRANSCODER_KEEPALIVE_PERIOD*3)     // in seconds

#define STREAM_ACTIVITY_TIMEOUT         TRANSCODER_KEEPALIVE_TIMEOUT        // in seconds

OR 

#define TRANSCODER_KEEPALIVE_PERIOD     2                                   // in seconds
#define TRANSCODER_KEEPALIVE_TIMEOUT    (TRANSCODER_KEEPALIVE_PERIOD*2)     // in seconds

#define STREAM_ACTIVITY_TIMEOUT         TRANSCODER_KEEPALIVE_TIMEOUT        // in seconds
------------------------------------------------------------------------------------------
in cstream.cpp

    // anything coming in from codec client ?
    if ( m_Socket.Receive(&Buffer, &Ip, 20) != -1 )  change from 1 to 20,

73... iz1mlt Luca

ea5gvk, 2018-12-06:
Hi, Thx Luca.
Are these modifications made before compiling the xlxd or can they be modified after compilation?

lucamarche-iz1mlt, 2018-12-06:
First you need to edit and then rebuild by running make clean and after make.

ea5gvk, 2018-12-06:
ok, Many thx. I try it, and I tell him about the experience.
A greeting.

lucamarche-iz1mlt, 2018-12-06:
Tnx! Wait the reports

73... Iz1mlt

ea5gvk, 2018-12-06:
Sorry, In original Code en Main.h
// Transcoder server --------------------------------------------

#define TRANSCODER_PORT                 10100                               // UDP port
#define TRANSCODER_KEEPALIVE_PERIOD     5                                   // in seconds
#define TRANSCODER_KEEPALIVE_TIMEOUT    30                                  // in seconds
#define TRANSCODER_AMBEPACKET_TIMEOUT 400 // in ms

modified, according to its indications it would be like this.

// Transcoder server --------------------------------------------

#define TRANSCODER_PORT                 10100                               // UDP port
#define TRANSCODER_KEEPALIVE_PERIOD     1                                   // in seconds 
#define TRANSCODER_KEEPALIVE_TIMEOUT    3                                  // in seconds  (1 * 3)
#define TRANSCODER_AMBEPACKET_TIMEOUT 1000 // in ms   ( 1 * 1000)

It is correct?

lucamarche-iz1mlt, 2018-12-06:
yes! 
Alternatively, I also have this second solution which is perhaps better than the first.
````
#define TRANSCODER_KEEPALIVE_PERIOD     2                                   // in seconds
#define TRANSCODER_KEEPALIVE_TIMEOUT    (TRANSCODER_KEEPALIVE_PERIOD + 1)     // in seconds
#define TRANSCODER_AMBEPACKET_TIMEOUT   (TRANSCODER_KEEPALIVE_PERIOD * 1000)  // in ms
````
The important thing is that AMBED_OPENSTREAM_TIMEOUT = STREAM_TIMEOUT because the problem solution is there. 
I chose this solution in main.h in order to have all the timers synchronized.

Find the code with the changes also on my repository.
https://github.com/lucamarche-iz1mlt

mlt

ea5gvk, 2018-12-08:
Many thx.

sm0tsc, 2018-12-08:
> yes!
> Alternatively, I also have this second solution which is perhaps better than the first.
>
> ```
> #define TRANSCODER_KEEPALIVE_PERIOD     2                                   // in seconds
> #define [...]

Does the forked XLXD work with  your fork of AMBEd with TEAM6160 support?

//TSC

ea5gvk, 2018-12-08:
Yes Luca. I compiled you github
https://github.com/lucamarche-iz1mlt/xlx_ambed --> For my Ambe BAOFARM1 (3 x Unit) 12 Stream
and
https://github.com/lucamarche-iz1mlt/xlxd  --> For my XLX051
Working Fine until now.

sm0tsc, 2018-12-11:
Luca after installing your branch on XLX010 with a remote ambed it seems we only get the callsign info no voice streams Please advise

//SM0TSC

ea5gvk, 2018-12-11:
Yes, correct.  At the time of being working, the audio is cut and only indicative output.
Please help check the code.

lucamarche-iz1mlt, 2018-12-11:
Strange! Aside from changing the timeouts and removing worning from compiling did not do anything else. What is written on the logs when this phenomenon happens?
I do however have a recheck to the code

sm0tsc, 2018-12-11:
> Strange! Aside from changing the timeouts and removing worning from compiling did not do anything else. What is written on the logs when this phenomenon happens?
> I do however have a recheck to the code

Nothing strange, no missed packets etc on AMBED side...
/TSC

lucamarche-iz1mlt, 2018-12-11:
> 
> 
> > Strange! Aside from changing the timeouts and removing worning from compiling did not do anything else. What is written on the logs when this phenomenon happens?
> > I do however have a recheck to the code
> 
> Nothing strange, no missed packets etc on AMBED side...
> /TSC

Confirm that the changes made to my fork work?
I seemed to understand that there were problems.
 
IZ1MLT

sm0tsc, 2018-12-11:
> Confirm that the changes made to my fork work?
> I seemed to understand that there were problems.
>
> IZ1MLT

Fork compiled 3 days ago does NOT work

//TSC

lucamarche-iz1mlt, 2018-12-11:
if you use the hardware boafarm you have to use the team6160 ambed, xlxd is fine regardless of the ambe card.

Ambed of the team6160 is also work with the other ambe hardware.
Try to believe, damage is not done, when it comes back.

If you have problems, put the output prints with the errors found.

IZ1MLT

ea5gvk, 2018-12-15:
Ok, Many thx, Luca, NOW yes operational. PERFECT. 
xlx051 and baofarm1 x 3 =12 stream, ALL PERFECT. 
MANY THX LUCA IZ1MLT.

lucamarche-iz1mlt, 2018-12-15:
Very well!

For news, I'm working on a version with read configuration from file, like jonathan software. I'll let you know.

73...  Luca

citynetcomputing, 2020-12-28:
@lucamarche-iz1mlt Has this timeout update fix been added into xlxd by default or do we still have to add it in ourselves? Also can you give the status on the configuration file version?

73

Bradn8pc, 2024-03-22:
I have your ambed file in and working it detects both my NW Digital thumbdv's and xlxd is loaded but I still get to transcoding going on.  there is no error on the ports.  it shows I have 2 channels with the 2 sticks.  what am I missing?

Bradn8pc, 2024-03-22:
By the way you can not get the 1.4.6 or the 1.4.8 drivers from ftdi under the linux directory as it has changed. can we please have an update as all the docs and files are 4 to 7 years old and outdated.