LX3JL/xlxd #117: AMBED with Flo DF3ET AMBE3003 and DV-30 Stick

hb9gfx, 2019-01-26:
As soon as I add a DV-Stick 30 Stick to the USB Port of the AMBE Server the AMBE3003 stops workling.
Both Ambes are recognized but no channels available.
As soon as I reject the DV-Stick 30 all is ok again.

phl0, 2019-01-26:
I could not check it yet but I guess the code fails pairing the spare AMBE channel of the DF2ET AMBE 3003 board with the channel of the DV-Stick 30. I have to take a look at the code.

hb9gfx, 2019-02-03:
Ist it an issue or just not supported

dl5di, 2019-03-25:
I found the same issue and proposed a change today to get it work.
I believe that there are actually 2 different issues:
1. There is a minor bug which lets the code run into a deadlock when you connect 2 sticks.
That is easy to fix.
2. Even when that is fixed the code does not really support to combine AMBE3000 with 3003.
But with a little change and a special treatment of this mixed configuation it works fine here at XLX262.
I could only test with the specific combination of one 3003 and one 3000, have no others.

73 de Hans DL5DI

phl0, 2019-03-26:
Hi Hans,

thanks for the info. I was also trying to debug this. Could I see you patch please?

   vy73 de Florian DF2ET

dl5di, 2019-03-26:
Uhps, sure.

I am no specialist on GitHub and thought that it would be visible at the
correct place automatically.
Maybe I did something wrong when recommending it.
The changed source file is located in this forked branch:

https://github.com/dl5di/xlxd/blob/patch-1/ambed/cvocodecs.cpp

73 de Hans DL5DI

phl0, 2019-03-26:
Hi Hans,

no problem. I could also have had a look at your repo :-)
I will give it a try.

   vy73 de Florian DF2ET

dl5di, 2019-03-26:
Last night I saw 2 branches in addition to the master in the xlxd repo.
One was older, the other showed my recommended changes.
Both are gone now, I only see the master branch left.

phl0, 2019-03-26:
Maybe you looked at the original repo rather than your fork. I can still see the changes in your forked repo: https://github.com/dl5di/xlxd/tree/patch-1?files=1

phl0, 2019-03-26:
First test:

```
~/xlxd/ambed $ !!                                                                                         
sudo ./ambed 10.99.0.156                                                                                                
Starting AMBEd 1.3.3                                                                                                    
                                                                                                                        
Initializing vocodecs:                                                                                                  
Detected 2 USB-FTDI devices                                                                                             
                                                                                                                        
Description : DF2ET-3003         Serial : AI041BTL                                                                      
Description : USB-3000   Serial : AI041BTM                                                                              
Opening DF2ET-3003:AI041BTL device                                                                                      
Trying DF2ET-3003 soft reset                                                                                            
DF2ET-3003 soft reset succeeded                                                                                         
ReadDeviceVersion : 0AMBE3003F1V120.E100.XXXX.C106.G514.R008.A0030608.C0020208/                                         
                                                                                                                        
                                                                                                                        
Opening USB-3000:AI041BTM device                                                                                        
FTDI function FT_Write error 0                                                                                          
FTDI function FT_Write error 0                                                                                          
FTDI function FT_Write error 0                                                                                          
FTDI function FT_Write error 0                                                                                          
FTDI function FT_Write error 0                                                                                          
FTDI function FT_Write error 0                                                                                          
FTDI function FT_Write error 0                                                                                          
FTDI function FT_Write error 0
```

Need to check my hars/soft reset settings on the USB-3000 board ...

dl5di, 2019-03-26:
I also found some open issues to fix with my modified old ThumbDVs,
which were for RPI/GPIO, but now modified for USB/FTDI.
But these things should not have anything to do with the detection and
pairing fix.
This stuff was just invisible before, because the device was not used at
all.
The worst was, that the ambed always run into a deadlock as soon as I
plugged in 2 sticks.

Have fun with fixing the other open issues! :-)

73 de Hans DL5DI

hb9gfx, 2019-07-04:
any news on this?

LX1IQ, 2019-12-02:
Hi folks, 

AMBED with Flo DF3ET AMBE3003 and DV-30 Stick should be fixed with the latest AMBEd release v2.3.1

73, Luc

closed 2019-12-02 by LX1IQ

phl0, 2019-12-02:
Jupp, I have tested latest changes by LX1JL successfully. Thank you for your support guys.

LX1IQ, 2019-12-02:
Tnx Flo, you're welcome.

73, Luc