LX3JL/xlxd #123: Stream close frames are reflected back to sender

nonoo, 2019-03-26:
The stream is closed right after a close frame is received and this unsets the associated source station of the stream. Right after the close the close frame is sent to all stations which are subscribed to the stream, including the source station, so it receives it's own frame.

sp2ong, 2020-06-17:
I confirm this problem and still exist in latest version xlxd

K2IE, 2020-11-02:
What would the appropriate fix be?  Would it be to not send the close frame to the source but to the other subscribed stations?

nonoo, 2020-11-02:
Yes

yo2loj, 2020-11-02:
The appropriate fix would be not to send ANY frame back to the source.

K2IE, 2020-11-06:
@yo2loj could I please trouble you to propose a fix that will work witht the current codebase?  I would be willing to give it a try on my reflector.

N5QM, 2020-12-09:
Guys is there any additional feedback or available workarounds for this issue?

LX1IQ, 2020-12-09:
Hi,

We are trying to publish a fix probably end of the year.
73, Luc

N5QM, 2020-12-09:
Thank you Luc, if you need any testers, please let me know and I will do my best to help out.

LX1IQ, 2021-01-05:
fixed in v2.4.1

73, Luc

closed 2021-01-05 by LX1IQ