LX3JL/xlxd #125: Feature Request: Outbound DPLus Connections

K2DLS, 2019-06-09:
TL;DR
It would be useful for xlxd to allow outbound DPLus connections for connection to REFxxx reflectors.

The traditional DPLus reflectors were designed to be a closed world.  Some of the dxrfd implementations seem to allow outbound linking via DPlus and I use this to link to an REF reflector on 2 channels concurrently.  It works fine.  But xlxd is the future.  It would be helpful to me to enhance xlxd with the capability to initiate outbound linking to a DPlus based REF reflector.  Multiple channel links should be supported.  As an example, on my existing dxrfd I link two channels to a DPlus reflector.

This feature could also be useful in overcoming the difficulty posed in initiating two concurrent outbound DExtra links between the same pair of reflectors on different modules.  DPlus could be used instead.

The power of xlxd is in its ability to bridge between multiple protocols.  Extending this bridge to legacy platforms as well would simplify my reflector infrastructure.

yo2loj, 2019-06-09:
This will probably be opposed by ref operators, because they will loose control on who connects another reflector to their network, since there is no distinction between a regular client and a xlx link.

K2DLS, 2019-06-09:
As I already mentioned, I am currently doing this with the version of dxrfd that I am using.  I would rather do it with xlxd and not have to use an older unmaintained piece of software.  I wonder how many REF only operators subscribe to this project anway?

yo2loj, 2019-06-09:
There are xlx forks that support xrf peers. Maybe that could be an answer :-)

K2DLS, 2019-06-09:
I have read #85 and I support integration of that into xlxd provided that it works reliably.  However, that does not satisfy what I am looking for, and that is an outbound connection via DPlus to an AA4RC DPlus reflector.  Unless you are saying that #85 does support DPlus outbound as well...

I wonder why #85 has not been accepted.  It seems useful.

yo2loj, 2019-06-09:
I have it running on my reflector for months without issues.
But no, it does not allow dplus links.
And it does not support multiple links to the same remote host.

vk4tux, 2019-06-09:
DPlus does not welcome this type of connection, and I doubt this will be 
supported in master xlxd branch.

It shows a lack of respect for the connect policy to DPlus reflectors, 
and attempts to camouflage external reflector

connections. This will lead to AA4RC banning the perpetrators on dplus 
systems, and possibly jeopardise the hotspot connect

protocol to dplus we already have. I say No, and i hope others say No to 
this ridiculous proposal also.

yo2loj, 2019-06-16:
You are right. But not all interconnects potentially using the DPlus protocol necessarily target a DPlus reflector. Any multiconnect without the full mesh requirements of the xlx interconnect protocol would be useful.

LX1IQ, 2019-12-02:
Hi folks,

This was already discussed a few times in the past. The XLX will not make any call outs, other than the buildin interlink function.

73, Luc

closed 2019-12-02 by LX1IQ