LX3JL/xlxd #128: D-Star traffic fails if transcoder fails

yo2loj, 2019-06-27:
I have seen this issue twice.
I have 2x DVsick33 setup as a transcoder in my home, while the actual reflector sits in a data center.
Sometimes the transcode to accept the transcoding, entering some kind of fail state. On the transcoder side, the streams show up, but there is an error on the XLX side.

Under these conditions, only the D-Star headers are sent by the XLX, while the actual DV stream is missing (at least in Dextra and Dplus). This manifests itself as nothing to be heard, with a short appearance of the callsign data at the end of transmission.

I haven't had the time to pursue this issue deeper, but shouldn't the reflector function normally and just ignore transcoding failures, while keeping up regular D-star traffic?

sm0tsc, 2019-06-28:
I experience the same.. I have not logged it yet.. Setup is the same with XLX server in Datacenter and ambeD on another site...
//SM0TSC

ik5xmk, 2019-12-07:
Hi, 

when you have this:
Jun 24 01:20:10 vps62907 xlxd: ambed 5713 of 2808 packets timed out

so **(X packets) > (Y packets) timed out** you need to restart AMBE server side (not xlxd). No other way i know. Is it possible to implement something in the code that does this automatically? Otherwise, transcoding is blocked, creating inconvenience between the various connected protocols.

Thank you, 73
David IK5XMK

vk4tux, 2019-12-07:
Just setup a watchdog script for it monitoring the ambed log.

This would never happen on a localhost system with DVSI.

ik5xmk, 2019-12-07:
I'm not sure that with a totally local system the problem can be definitively managed. I have tried with various ThumbDV and DV3x USB dongles also on the same server and now with two Baofarm cards.
This loss of synchronization can also occur with meetings of different protocols, overmodulation, bad client data connections, etc.
Log control is certainly a viable method, but perhaps a control protocol between xlx and its ambe server could also be used, regardless of whether it can be installed. 
Thanks, I try this suggestion. Many 73.

vk4tux, 2019-12-07:
I use 2 x 3006 and 2 x dvstick33, all genuine DVSI 3003 1MBaud rate 
chips,on a local host system here with PC server for XLX389 with usb 
transcoders attached.

I never had your issue with ambed.

root@XLX389:/var/www/db/pgs# cat /ambed/run
#!/bin/bash
# start ambed server

sudo rmmod ftdi_sio
sudo rmmod usbserial
sudo /ambed/ambed 127.0.0.1 >/var/log/ambed.log &

so with ambed.log you can monitor ;

You can setup /usr/bin/ambechk.sh and run every minute from root crontab ;

..................................................................................................................

#!/bin/bash
sleep 5
PATTERN=timed
FILE=/var/log/ambed.log
if grep -q $PATTERN $FILE;
  then
      echo "ambed timeout error"
      > /var/log/ambed.log
      /usr/sbin/service ambe restart <<<<<<<<<<<< replace with your ambe 
start script/service etc..
  else
      echo "all ok"
      echo "Exiting..."
      exit 0
fi

................................................................................................................

Adrian ... vk4tux

vk4tux, 2019-12-08:
Yes, well you can't beat a localhost connection, but I know this is not
always possible, but every effort should be made to use a physical hardware
server.