LX3JL/xlxd #129: Feature Request: listening on interface instead of IP

iw5ejm, 2019-07-18:
Is possible to bind the server to a specific interface instead of an ip?

This could allow clients from different subnet to connect xlxd.

I have regular client like peers from internet ad other (some repeater) reacheable only via LAN (wifi ham network).

iw5ejm, 2019-07-18:
> use iproute2 to configure routes properly. It's enough

How sorry? The concept I mean.
The ubuntu server  has the proper set of route rules, but still xlxd listen
only on the class where is launched.

Email.it Ze-FREE: l'unica casella GRATUITA su piattaforma professionale licenziata Zimbra Network Edition. Registrati qui: http://bit.ly/reg-email-free
---
Email.it Ze-Free: the only FREE email account hosted on a Zimbra Network Edition lincensed professional platform. Sign in here: http://bit.ly/reg-email-free

yo2loj, 2019-07-18:
Have you tried using 0.0.0.0 as the xlx bind address? It actually works...

Email.it Ze-FREE: l'unica casella GRATUITA su piattaforma professionale licenziata Zimbra Network Edition. Registrati qui: http://bit.ly/reg-email-free (http://bit.ly/reg-email-free)
---
Email.it Ze-Free: the only FREE email account hosted on a Zimbra Network Edition lincensed professional platform. Sign in here: http://bit.ly/reg-email-free (http://bit.ly/reg-email-free)

iw5ejm, 2019-07-18:
> Have you tried using 0.0.0.0 as the xlx bind address? It actually works...

Thanks man!! God bless you... it was so simply HI HI 
Columbus' egg we call it in italy!

I think that with 1.* versions this trick doesn't work. 
I seem to have tried with that version. However... With 2.2.2 works like a charm!!

73's!!!

closed 2019-07-18 by iw5ejm

cyanide-burnout, 2019-07-18:
Uuuh, initially i was thinking you are trying to route traffic over several interfaces with different IPs... Because 0.0.0.0 is so obvious... sorry.