LX3JL/xlxd #132: YSFprotocol main.h option for default module on connect.

vk4tux, 2019-10-28:
The more I test the YSF feature, the more I desire an option in main.h to just connect YSF client connections to a defined module in main.h. I cannot believe this is not possible.

yo2loj, 2019-10-28:
I think Luc follows the same paradigm as in DMR.
The system is completely autonomous and self contained.
In DMR, one needs to use 4001 - 4026 to reach a specific module, In YSF, one uses Wires-X commands to connect to a specific module.
But a default module would make a lot of sense.

vk4tux, 2019-10-28:
It could be a sysop advanced option available in main.h. For YSF2DMR 
bridge this is ideal.

yo2loj, 2019-10-28:
I would go a step further: Allow to name and register that default module in the YSF registry...
But maybe that is just wishful thinking :-)

LX1IQ, 2019-10-28:
Let's wait what the OpenSpot and Pi-star guys are doing.
I'm in touch with Norbert and he will probably add the DMTF command to the OpenSpot2 for a default Linking option to a specified Module.

> I think Luc follows the same paradigm as in DMR, as the system is completely autonomous and >self contained. 
@ Marius, absolutely.

LX1IQ, 2019-10-28:
@ Adrian, setting any default YSF module via the main.h does not make any sense to me, except you would like to link other systems to the XLX without using the XLX Interlink protocol. Like HBlink or what ever.

vk4tux, 2019-10-28:
There are several reasons, to direct connections to where fusion is 
preferred. We have two inhouse wires-x links and a hblink system built 
within xlx389 on same server.

I gave up xlxd.interlink some time back, and also have no interest in 
brandmeister , connecting to it or using their systems. So we would like 
default YSF connections on Q module.

It will work well. However just as a option in main.h, as that is were 
other options are for the compilation.

I am a time intensive big user and system sysop here, always looking for 
improvements, and for our application and connections to wires-x, it 
makes perfect sense to have and use that option here.

You will find that many like me will have use for it.

DigiDaz, 2019-11-16:
Does anyone know which file to modify to connect to a module by default? This may allow me to connect users until I can find out how to properly select a module. I have an experienced Openspot2 user finding it impossible at the moment.

LX1IQ, 2019-11-17:
Install XLX v2.3.2.
This version allows you to define a default YSF module which your clients are linked to.

73, Luc

closed 2019-11-17 by LX1IQ