LX3JL/xlxd #133: How to disable YSF on xlxd v2.3.1

davidcien, 2019-11-11:
Dear All
i had ysf reflector already ,now i want to install xlxd but the new version 2.3.1 include ysf 
how can i disable it ? 
thanks a lot.

vk4tux, 2019-11-11:
Hi David, I have the answer, but are you receiving my emails ? You did 
not respond regarding ircddbgateway      ?

Adrian ... vk4tux

LX1IQ, 2019-11-11:
Hi David,
There is no way to disable YSF. The only difference between v2.3.1 and 2.2.2 is the YSF implementation.
In addition, the ysf master host file on the API server is generated based on the version number.
If you run v2.3.1 you should not block port 42000 !
What you can do is to change your YSF port to something other than port 42000. So you will run XLX YSF on port 42000 and your YSF on 42001 (for example). Both server could run on the same machine.
73, Luc

davidcien, 2019-11-11:
Sorry Ircddbgateway install success(2015 version on pi ) ,thanks you info 
Now I try to install xlx  on Google cloud ,it already  ysf reflector  running ,then I found xlx had new version on github with ysf install
Please tell me how to disable it , Thanks a lot 
傳送自 Yahoo奇摩電子信箱 Android 版 
 
   在 2019 年 年 11月 月 11 日 日週一，時間：14:13 ， Adrian Fewster <notifications@github.com> 寫道：   Hi David, I have the answer, but are you receiving my emails ? You did 
not respond regarding ircddbgateway      ?

Adrian ... vk4tux

vk4tux, 2019-11-11:
Ok, This earlier version has no YSF ;

https://github.com/LX3JL/xlxd/archive/8885b722956d5296edbbc9a8b39613ede26655dd.zip

davidcien, 2019-11-11:
thank a lot

 

https://github.com/LX3JL/xlxd/archive/8885b722956d5296edbbc9a8b39613ede26655dd.zip

xe1dvi, 2019-11-15:
What are the advantages of having YSF capability?
Can you connect directly via YSF?? and how???
I have a YSF reflector already linked to my XLX reflector, will this new version will link it together evn if they are on the same server??
Can I change YSF port on XLX reflector instead of on YSF reflector??
I know too many questions at once, but I just need answers please.

Thank you in advance.
Rey
W6DVI

vk4tux, 2019-11-15:
It works direct embedded in xlxd 2.3.1 with default port 42000 UDP.

Once connected, wires-x is required by the user to select the module 
connect. See XLX389 dashboard for config example and wires-x videos etc.

There is a big general sysop desire to have a default module autoconnect 
on initial YSF connect, but we are still trying to convince Luc to add 
it, so a default module connect happens

on first connect. Then the user can change with wires-x, with no auto 
revert to default. pistar users will find this beneficial and the XLX 
sysop and direct YSF traffic where it is most suited.

The xlx ysf database xlx list could be scripted into the YSFHost file by 
Kim DG9VH, in Germany or by the pistar authors.

xe1dvi, 2019-11-15:
I really appreciate your detailed information, thanks to you I have all figured out.
Thank you again!!!
Rey.
W6DVI

davidcien, 2019-11-15:
Thanks 

傳送自 Yahoo奇摩電子信箱 Android 版

DigiDaz, 2019-11-16:
My users are unable to use the wires-x from things like Openspot2.

They show connected on the dashboard with YSF protocol but the module column is empty.

Pressing the wires-x button on the radio does nothing.

Also https://register.ysfreflector.de/ reports no response to pings.

LX1IQ, 2019-11-17:
Hi DigDaz,

XLX is not a YSFReflector, more an YSF Master which provides 26 YSF rooms.
The XLX is not responding to the ysfreflector register.
With XLX version 2.3.2 you can predefine a default module which will your client linked to.

I hope it helps.
73, Luc

closed 2019-11-17 by LX1IQ