LX3JL/xlxd #141: Multiple Ambed streams fail!

citynetcomputing, 2019-12-17:
Hello,

I have ambed running with xlxd. During one stream it works fine. If another stream starts both fail and i have to stop ambed for a bit and restart it. I have 4 dongle that all show up fine. I compiled both ambed and xlxd from src. Any ideas why I cant run parallel streams??

Thanks!

vk4tux, 2019-12-17:
Are you running ambe hardware on server, or remote   ?

citynetcomputing, 2019-12-17:
All same server. <div>
</div><div>
</div><div>-------- Original message --------</div><div>From: Adrian Fewster <notifications@github.com> </div><div>Date: 12/16/19  10:05 PM  (GMT-05:00) </div><div>To: LX3JL/xlxd <xlxd@noreply.github.com> </div><div>Cc: Jay <jay@cidcomm.com>, Author <author@noreply.github.com> </div><div>Subject: Re: [LX3JL/xlxd] Multiple Ambed streams fail! (#141) </div><div>
</div>Are you running ambe hardware on server, or remote   ?

vk4tux, 2019-12-17:
Do you have any log showing the failure, and transcoder hardware type ?

LX1IQ, 2019-12-17:
...  I compiled both ambed and xlxd from src. Any ideas why I cant run parallel streams??
Are you using two ambed servers with one XLX?
Your 4 dongles needs to be on one ambed server.

73, Luc

citynetcomputing, 2019-12-17:
No, single xlxd server with ambed running locally on it with 4 dongles attached.<div>
</div><div>
</div><div>-------- Original message --------</div><div>From: LX1IQ <notifications@github.com> </div><div>Date: 12/17/19  12:49 AM  (GMT-05:00) </div><div>To: LX3JL/xlxd <xlxd@noreply.github.com> </div><div>Cc: Jay <jay@cidcomm.com>, Author <author@noreply.github.com> </div><div>Subject: Re: [LX3JL/xlxd] Multiple Ambed streams fail! (#141) </div><div>
</div>...  I compiled both ambed and xlxd from src. Any ideas why I cant run parallel streams??
Are you using two ambed servers with one XLX?
Your 4 dongles needs to be on one ambed server.

73, Luc

citynetcomputing, 2019-12-17:
I was running on VMWare/Debian. I just moved everything over to a bare bones server with 16 CPUs and 256GB of ram. We will see how she does on that ;) Hopefully it works better. I'm running 4 of the NWDigital Thumb DV's on ambed.

Dec 17 13:57:37 XLX891 xlxd: DMRmmdvm configuration packet from N2CID   B at 174.141.222.67
Dec 17 13:57:37 XLX891 xlxd: DMRmmdvm client N2CID   B linking on module C
Dec 17 13:57:38 XLX891 xlxd: DExtra connect packet for module D from W4END   B at 108.93.65.137 rev 0
Dec 17 13:57:38 XLX891 xlxd: New client W4END   B at 108.93.65.137 added with protocol DExtra on module D
Dec 17 13:57:38 XLX891 xlxd: DExtra connect packet for module D from W4END   B at 108.93.65.137 rev 0
Dec 17 13:57:43 XLX891 xlxd: DExtra connect packet for module C from N2CID   B at 174.224.137.163 rev 0
Dec 17 13:57:43 XLX891 xlxd: New client N2CID   B at 174.224.137.163 added with protocol DExtra on module C
Dec 17 13:57:43 XLX891 xlxd: DExtra connect packet for module C from N2CID   B at 174.224.137.163 rev 0
Dec 17 13:57:48 XLX891 xlxd: ambed openstream ok
Dec 17 13:57:48 XLX891 xlxd: Opening stream on module D for client BM3103   with sid 53808

citynetcomputing, 2019-12-17:
Here's ambedtest also on new system.

Stream 51 (1->2) : 100 / 99 / 0 : 49.5 fps
Stream 52 (1->2) : 100 / 0 / 0 : 0.0 fps
Stream 49 (2->1) : 100 / 101 / 0 : 50.5 fps
Stream 50 (2->1) : 100 / 0 / 0 : 0.0 fps
Stream 51 (1->2) : 100 / 101 / 0 : 50.5 fps
Stream 52 (1->2) : 100 / 0 / 0 : 0.0 fps
Stream 49 (2->1) : 100 / 99 / 0 : 49.5 fps
Stream 50 (2->1) : 100 / 0 / 0 : 0.0 fps
Stream 51 (1->2) : 100 / 101 / 0 : 50.5 fps
Stream 52 (1->2) : 100 / 0 / 0 : 0.0 fps
Stream 49 (2->1) : 100 / 100 / 0 : 50.0 fps
Stream 50 (2->1) : 100 / 0 / 0 : 0.0 fps
Stream 51 (1->2) : 100 / 98 / 0 : 49.0 fps
Stream 52 (1->2) : 100 / 0 / 0 : 0.0 fps
Stream 49 (2->1) : 100 / 100 / 0 : 50.0 fps
Stream 50 (2->1) : 100 / 0 / 0 : 0.0 fps
Stream 51 (1->2) : 100 / 101 / 0 : 50.5 fps
Stream 52 (1->2) : 100 / 0 / 0 : 0.0 fps
Stream 49 (2->1) : 100 / 100 / 0 : 50.0 fps
Stream 50 (2->1) : 100 / 0 / 0 : 0.0 fps
Stream 51 (1->2) : 100 / 101 / 0 : 50.5 fps
Stream 52 (1->2) : 100 / 0 / 0 : 0.0 fps
Stream 49 (2->1) : 100 / 101 / 0 : 50.5 fps
Stream 50 (2->1) : 100 / 0 / 0 : 0.0 fps
Stream 51 (1->2) : 100 / 99 / 0 : 49.5 fps
Stream 52 (1->2) : 101 / 0 / 0 : 0.0 fps
Stream 49 (2->1) : 100 / 100 / 0 : 50.0 fps
Stream 50 (2->1) : 100 / 0 / 0 : 0.0 fps
Stream 51 (1->2) : 100 / 100 / 0 : 50.0 fps
Stream 52 (1->2) : 100 / 0 / 0 : 0.0 fps
Stream 49 (2->1) : 100 / 99 / 0 : 49.5 fps
Stream 50 (2->1) : 100 / 0 / 0 : 0.0 fps
Stream 51 (1->2) : 100 / 101 / 0 : 50.5 fps
Stream 52 (1->2) : 100 / 0 / 0 : 0.0 fp

citynetcomputing, 2019-12-17:
And more Ambed startup info..

Initializing vocodecs:
Detected 4 USB-FTDI devices

Description : DVstick-30         Serial : DM42SHZ8
Description : DVstick-30         Serial : DM4C1PHC
Description : DVstick-30         Serial : DM4A713L
Description : DVstick-30         Serial : DM4A714P
Opening DVstick-30:DM42SHZ8 device
ReadDeviceVersion : 0AMBE3000R1V120.E100.XXXX.C106.G514.R009.B0010411.C0020208

Opening DVstick-30:DM4C1PHC device
ReadDeviceVersion : 0AMBE3000R1V120.E100.XXXX.C106.G514.R009.B0010411.C0020208

Opening DVstick-30:DM4A713L device
USB-3000 soft reset failed

Codec interfaces initialized successfully : 2 channels available

Initializing controller

AMBEd started and listening on 127.0.0.1

citynetcomputing, 2019-12-18:
Yea so its still doing it on the new system. Audio will drop out randomly on one stream or the other :(

vk4tux, 2019-12-18:
Just to let you know ; DVstick-30:0 are not NW Digital Thmbdv's as you said, they are DVMega products from Guus.You did not comment on whether you cleaned the usb connections. The reset failure happens often when these connections are ordinary, even with the dvstick33. I take mine out and contact clean regularly for faultless performance. Also you did not mention if you are using USB 3.0 or USB 2.0 ports ?

citynetcomputing, 2019-12-18:
Hey all,

So i applied the fix's from here - https://github.com/LX3JL/xlxd/issues/115 which seem to have resolved most of it from what I can tell. Not getting any errors or drops and we had both streams running seemingly perfect last night minus a few hickups..

The dongles I am using are these - https://www.gigaparts.com/dvmega-dvstick-30.html

I did not clean the USB connections as its a brand new server and new DV-Sticks. Also the ports are USB3 but don't know off hand if debian has the drivers for USB3. Also the server is in a Tier4 Data Center which would make any frequent USB port cleaning problematic ;)

Can anyone post what settings they are using in reference to the above linked issue? Maybe I can tweak mine a little better... ;)

vk4tux, 2019-12-18:
Many times new USB products have a coating on the metal not conductive, 
which relies on the rubbing of the insert to break through for best 
conduction.

On Odroid devices the cleaning of USB ports is recommended by the 
manufacturer for the same reason.

USB2.0 ports will give you a better result on dvstick or thumbdv's.

citynetcomputing, 2019-12-19:
> Many times new USB products have a coating on the metal not conductive, which relies on the rubbing of the insert to break through for best conduction. On Odroid devices the cleaning of USB ports is [...]

From an electrical standpoint this makes no sense as the outer metal portion is for support only. Its non-electrical. Only the 4 copper pins inside are actually used for Gnd/TX/Rx etc. But hey, whatever works ;)

citynetcomputing, 2019-12-19:
Looks like we are running at USB2 speeds also so... I also changed the buffer in ambed from 1, then 20, then back 0 which seems to work best.

Device Descriptor:
  bLength                18
  bDescriptorType         1
  **bcdUSB               2.00**
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0         8
  idVendor           0x0403 Future Technology Devices International, Ltd
  idProduct          0x6015 Bridge(I2C/SPI/UART/FIFO)
  bcdDevice           10.00
  iManufacturer           1 FTDI
  iProduct                2 DVstick-30
  iSerial                 3 DM4C1PHC
  bNumConfigurations      1

citynetcomputing, 2019-12-20:
What do these stats fields mean. Cant find any info on it. ping times?
 ambed stats (ms) : 0.0/32.1/60.5

citynetcomputing, 2019-12-23:
is the 160.7 normal or? Don't really know what these stats mean...

dmr xlxd: ambed stats (ms) : 0.0/29.2/160.7

citynetcomputing, 2019-12-25:
Hello All.

Still having issues running multiple streams together. Transcoding DMR with D-STAR using xlxd and ambed. One works fine. When running two occasionally i get these stats "Dec 24 21:31:44 dmr xlxd: ambed stats (ms) : 0.0/29.6/301.1" and ambed has to be restarted to clean it all up. Ive tried everything,

Different OS,s
Different Hardware,
Cleaning USB Ports
Upping CPU and RAM.
Barebones and VM Based.
Adjusted sync and timers etc
Used different versions of both and different repos

Same results on everything... Thoughts???

citynetcomputing, 2019-12-30:
> 
> 
> ... I compiled both ambed and xlxd from src. Any ideas why I cant run parallel streams??
> 
> It should be no problem to run parallel streams.
> On the XLX270 I have 12Ch transcoding. 6 parallel streams can run without trouble.
> 
> Did you run the ambedtest program ? Is your server supporting the load. What hardware are you using for transcoding ?
> 
> 73, Luc

Hi Luc,

Are you using multiple streams to BM via direct peer?