LX3JL/xlxd #145: Can make ambed support single AMBE3003 with single AMBE 3000 ?

bv3un, 2020-01-03:
Dear Sir
as i know AMBE 3003 is 3 channel and AMBE 3000 is one channl,in current ambed version that only supprt 2 channel on 3003 .is it fully support 3 channel in the future ? and support single 3003 + single 3000.
they can  use both to provide 4 channels transcode it will be great
thanks

LX1IQ, 2020-01-03:
Hi David,

Have a look at the AMBed server readme.
You can combine 1x 3003 + 1x 3000 to get 4 channels already.

73, Luc

DECHARGE

Ce message et toutes pièces jointes sont établis à l'intention exclusive de ses destinataires. Ils peuvent contenir des informations confidentielles. Si vous recevez ce message par erreur, merci de le détruire et d'en avertir immédiatement l'expéditeur. Toute utilisation de ce message non conforme à sa destination, toute diffusion ou toute publication, totale ou partielle est interdite, sauf autorisation expresse. Ce message a fait l'objet d'un traitement anti-virus.

Le contenu de ce message et des pièces jointes ne pourrait engager la responsabilité de l’AGEDOC asbl que s'il a été émis par une personne dûment habilitée agissant dans le strict cadre des fonctions auxquelles elle est employée et à des fins non étrangères à ses attributions.

bv3un, 2020-01-03:
Dear Sir
I saw in readme it is 
3003-3000(pair) not signal ,and I had try with my 3003 and 3000 together but still only 2 channels
Maybe I use old version of ambed not the 1.3.4 ,I will try to upgrade and test

傳送自 Yahoo奇摩電子信箱 Android 版 
 
   在 2020 年 年 1月 月 3 日 日週五，時間：18:58 ， LX1IQ <notifications@github.com> 寫道：   Hi David,

Have a look at the AMBed server readme.
You can combine 1x 3003 + 1x 3000 to get 4 channels already.

73, Luc

LX1IQ, 2020-01-03:
Hi,
The latest version of the AMBed Server supports 4 channels out of one USB-3003 and one USB-3000.
Only original AMBE Chips by DVSI are supported out of the box.

73, Luc

bv3un, 2020-01-04:
Dear SirHere is my report

1.After upgrade ambed  to last versionStill have two channel ,
escription : DVstick-33 Serial : Description : FT230X Basic UART Serial :Opening DVstick-33:DM47MSO3 deviceTrying DF2ET-3003 soft resetDF2ET-3003 soft reset succeededReadDeviceVersion : 0AMBE3003F1V120.E100.XXXX.C106.G514.R008.A0030608.C0020208/

Codec interfaces initialized successfully : 2 channels available
I am sure 3000 is DVSI ,but the description is FT230X Basic UART Serial not DVsitck-30 ,thus I go www.ftdichip.com/Support/Utilitirhtm to download FT-PROG ,and use it modified description to DVstick-30
Now. 4 channels are fine 
Thanks a lot 
傳送自 Yahoo奇摩電子信箱 Android 版

vk4tux, 2020-01-04:
I have seen this before here with dvstick30

http://xlxbbs.epf.lu/viewtopic.php?t=194

I cc to Guus, to help ensure desc is done at manufacture.

bv3un, 2020-01-04:
Hi Adrian Thanks a lot By the way ,may I know how to connect administrator of http://xlxbbs.epf.lu ? Because I had registered this site with 4 times and can not received any approved mail in half year,I try to use  yahoo or gamil and get the same results

傳送自 Yahoo奇摩電子信箱 Android 版 
 
   在 2020 年 年 1月 月 4 日 日週六，時間：9:39 ， Adrian Fewster <notifications@github.com> 寫道：   I have seen this before here with dvstick30

http://xlxbbs.epf.lu/viewtopic.php?t=194

I cc to Guus, to help ensure desc is done at manufacture.

vk4tux, 2020-01-04:
I would ask Luc on this mailing list for that.

LX1IQ, 2020-01-04:
Hi David,

Use your callsign as the username to register @ xlxbbs.

73, Luc

在 2020 年 年 1月 月 4 日 日週六，時間：9:39 ， Adrian Fewster <notifications@github.com<mailto:notifications@github.com>> 寫道： I have seen this before here with dvstick30

http://xlxbbs.epf.lu/viewtopic.php?t=194

bv3un, 2020-01-04:
Dear Sir 
Done de BV3UN 

傳送自 Yahoo奇摩電子信箱 Android 版 
 
   在 2020 年 年 1月 月 4 日 日週六，時間：16:02 ， LX1IQ <notifications@github.com> 寫道：   Hi David,

Use your callsign as the username to register @ xlxbbs.

73, Luc

http://xlxbbs.epf.lu/viewtopic.php?t=194

closed 2020-01-04 by LX1IQ