LX3JL/xlxd #152: No telemetry info on YSF

xe1dvi, 2020-02-13:
Hello,
I have a YSF repeater running on pi-star, if the repeater in connected to my YSF reflector telemetry passes thru and I can see the call sign and distance of stations on the other end as they can see mine.

When I connect the repeater to my XLX077 reflector I can only see  the call sign but not distance, telemetry is not passing thru.

I have the latest 2.3.5, maybe there is a setting that I need to activate, please advice.

Thank you.

vk4tux, 2020-02-13:
That is normal. XLX YSF connect is converted to DMR, then back to YSF 
for any other YSF client such that no GPS data is shared, as there is no 
YSF throughput.

There is nothing you can do about it.

Best to stay with YSFReflector, and use YSF2DMR back to XLX.

xe1dvi, 2020-02-13:
Ahhhhh good to know!!!
Yeah thats how I have it linked!
Thanks Adrian.

Sincerely,
Rey.XE1DVI / W6DVI

closed 2021-01-17 by LX1IQ