LX3JL/xlxd #154: YSF connection duplicates on nodes dashboard

iw2gob, 2020-02-26:
Hi,

the YSF connection is still active on the dashboard, despite the disconnection (after hotspot power off). If I make a new connection, duplicates are created on the dashboard; they disappear only after xlxd is restarted.

73, Maurizio IW2GOB

yo2loj, 2020-02-27:
Since the connect and keep alive messages in YSF are the same, this may only happen if the hotspot's IP address changed between connection loss (power off will not disconnect) and reconnect. Could you confirm this scenario?

LX1IQ, 2020-02-27:
How Maurizio,
If you power of your hotspot without a proper disconnect, the HotSpot is not able to send a disconnect packet to the XLX. Sou you have to wait a bit until the XLX disconnects that HotSpot via the keep alive mechanism. The time when that happened depends on what you did put in the main.h before compiling your XLX.

73, Luc

iw2gob, 2020-02-27:
Hi Luc and yo2loj,  I'm using openspot2 and openspot3 with YSF connector to XLX997.
if I turn off my openspot3, the YSF connection remains active on the XLX dashboard ... even if hours pass! In this case the keep alive packets should expire and the connection should disappear. If I re-turn on the openspot3 on the same connection (TIM fiber) for example after 1 hour, a new connection is created with the same IP. If I turn off and on again after 1 hour on a 4G connection, another YSF connection from another IP appears on the dashboard. All these YSF connections remain active in the dashboard and accumulate one after the other, even if coming from the same IP. If I restart the XLX service, they all disappear.
I don't think it depends on the Openspot, because if I turn it off it doesn't send the keep alive packets anymore ... and the connection should drop shortly after.
I hope I managed to explain myself

Thank you,

73, Maurizio IW2GOB

yo2loj, 2020-02-27:
I tried to reproduce the behavior on a OpenSpot 1 by disconnecting the power.
The node dissappeared in a few seconds and reconnected as expected, without issues.

iw2gob, 2020-02-27:
Hi Marius, are you trying with a YSF connection to XLX ? The problem is only with YSF connection to XLX. If you check now on my xlx997 http://xlx997.iw2gob.it, you will see 2 YSF connections active since 16:42 and 16:44....but my openSPOT3 is power off since 16:30. Why XLX is showing 2 YSF active connections ? The keep alive packets have expired for hours and these 2 connections had to be automatically removed by XLX or am I wrong?
Thank you,

Maurizio IW2GOB

iw2gob, 2020-02-27:
Sorry, my openSPOT3 is powered off since 16:50....

yo2loj, 2020-02-28:
Yes, I tried native YSF connection to XLX227C. Could you try to connect to that one and then power of you OS3 for a test?

iw2gob, 2020-02-28:
Hi Marius, I'm trying to connect to XLX227C with YSF, but your XLX version (2.3.3) doesn't support YSF status protocol and OS3 cannot connect; you will see IW2GOB-Y, but on my OS3 the connector is connecting....My XLX is latest version 2.3.5 and support YSF connection status protocol....with latest XLX version I'm having the problem I reported. 

73, Maurizio IW2GOB

LX1IQ, 2020-02-28:
Well try XLX270, it is on the latest version.
73, Luc

iw2gob, 2020-02-28:
Hi Luc, 
on the XLX270 I don't find the problem! So I recompiled my XLX997 and the problem disappeared. I think I made some mistakes in the compilation. I apologize for making you waste time. Thanks for the support.

73, Maurizio IW2GOB

LX1IQ, 2020-02-28:
No problem, you're welcome.
73, Luc

closed 2020-02-28 by LX1IQ