LX3JL/xlxd #156: DMRmmdvm packet (19)

mrnonaki, 2020-03-07:
my XLX520 flooding "DMRmmdvm packet (19)"
What does code 19 mean?
[image]

narspt, 2020-04-04:
I'm no expert but on a quick look at source: https://github.com/LX3JL/xlxd/blob/be87d548970804989d9f2bc4fb3b343fc3586d38/src/cdmrmmdvmprotocol.cpp#L249 it seems such message means there is an unhandled/unrecognized packet incomming on DMRmmdvm protocol port with size of 19 bytes (that 19 is not a code but length), maybe you should start by identifying if it is some specific DMR client connected on your XLX server that causes it? Or if you actually get that even with no DMR clients connected maybe it is something else probing the server on that port? (on this case you may try to identify source ip, etc using tcpdump and eventually block it on iptables...)

narspt, 2020-04-04:
I'm no expert on this but on a quick look at source: https://github.com/LX3JL/xlxd/blob/be87d548970804989d9f2bc4fb3b343fc3586d38/src/cdmrmmdvmprotocol.cpp#L249 it seems such message means there is an unhandled/unrecognized packet incomming on DMRmmdvm protocol port with size of 19 bytes (that 19 is not a code but length), maybe you should start by identifying if it is some specific DMR client connected on your XLX server that causes it? Or if you actually get that even with no DMR clients connected maybe it is something else probing the server on that port? (you may try to identify source ip, etc using tcpdump and eventually block it on iptables...)

LX1IQ, 2021-01-17:
We did not implement the embeddedLC.
This part of the protocol is not relevant for the XLX functionality.

73, Luc

closed 2021-01-17 by LX1IQ