LX3JL/xlxd #157: Upgrade XLX reflector to support ICOM Terminal mode

1erbn, 2020-03-09:
How can I upgrade my XLX reflector while maintaining the current configuration and the dashboard?
As I am not that experienced in Linux the answer to my question in https://github.com/LX3JL/xlxd/pull/140 made me doubt. Since I want to keep my reflector up and running, I am looking for a clear procedure to upgrade my reflector.
Does anyone want to help me?

Thanks.

yo2loj, 2020-03-09:
Unpacl the zip file from github or clone the project somewhere to your liking.
You need to compile your new version (command 'make' in the src directory of the sorce file), stop your reflector ('killall xlxd') and overwrite your xlxd file (wherever you installed it) with the one compiled in the src directory.
Restart your system and it should start using the new file.

1erbn, 2020-03-11:
This evening I updated my reflector and everything is working from the LAN side. Now I copied the xlxd.terminal file into place and put my external ip in it. As you said after that it's not working anymore from the LAN side. Are you able to test it to be sure it's working from the cloud? Address is xlx131.pe1er.nl
Thanks again for your support.

yo2loj, 2020-03-12:
Yes, it seems it is working as expected.

citynetcomputing, 2020-07-25:
Upgrade failed. any ideas??

Jul 24 22:56:46 xlx891 xlxd: Error opening socket on port UDP40000 on ip 209.251.60.221
Jul 24 22:56:46 xlx891 xlxd: Error opening socket on port UDP12346 on ip 209.251.60.221
Jul 24 22:56:46 xlx891 xlxd: Error opening socket on port UDP12345 on ip 209.251.60.221
Jul 24 22:56:46 xlx891 xlxd: Error opening raw socket for ICMP
Jul 24 22:56:46 xlx891 xlxd: Transcoder connected at 127.0.0.1
Jul 24 22:56:46 xlx891 kernel: [10933771.657836] traps: xlxd[30031] general protection ip:7f6d9fee60d4 sp:7ffc173bd960 error:0 in libstdc++.so.6.0.22[7f6d9fe2c000    +172000]

yo2loj, 2020-07-25:
These error appear usually if there is another xlxd instance already running on your system.
Make sure you stop the already running xlxd. A system restart resolves this by force.

citynetcomputing, 2020-07-25:
Wierd. I did a killall xlxd and it showed no instances but I replaced it and rebooted and seemed to work..

Thanks!!

closed 2020-07-27 by LX1IQ