LX3JL/xlxd #158: No YSF pass though.

2E0LXY, 2020-03-24:
Help please.
I've installed XLX on a Debain VPS, all was working well but, now there is no YSF audio and the dashboard does not show the "Audio" flashing icon.
My MMDV connects fine to is and YSF is registered and pings OK.
YSF44904 XLX899.
Fixed IP of 45.14.224.180

Many thanks

Daren
2E0LXY

yo2loj, 2020-03-24:
I just checked it.
There is no default module for connection on your XLX. That means that in order to connect, you need to activate the Dx mode, request the reflector list and connect to one of the modules. After that, the flashing audio icon appears, and also the connected callsign in the corresponding module.

2E0LXY, 2020-03-24:
Hi.
I have selected DX on my FT1XD and selected the XLX899 from the list but how do I connect to one of the modules please and how can I set a default module on the XLX please?
Thank you for your help :-)

yo2loj, 2020-03-25:
That is described in your user manual... But the short version:
- Do a long press on your 'Dx' button. The radio will send out a connect request and connect to the reflector
- Do a long press on 'ENT'. A menu will appear (something starting with C1: ________). Using the rotary button, scroll down to 'All' and press 'ENT' shortly. Wait for the reflector response.
- Using the rotary button, select your module and press 'ENT' again. The radio should connect to the module.
- Scroll to 'Back' and press 'ENT' to exit to the main screen.

To be able to set a default module, you need to enable it in the src/main.h file, by setting the option YSF_AUTOLINK_ENABLE to 1 and adapt your YSF_AUTOLINK_MODULE to your needs.

2E0LXY, 2020-03-28:
Thank you for your help.

I have chaged my MMDVM to UPPERCASE Hostfiles: and everything started to make sense!

If I have XLX already running how do I..

"To be able to set a default module, you need to enable it in the src/main.h file, by setting the option YSF_AUTOLINK_ENABLE to 1 and adapt your YSF_AUTOLINK_MODULE to your needs."

Many thanks for your time and answers.

Daren

closed 2020-04-04 by 2E0LXY