LX3JL/xlxd #161: YSF_AUTOLINK_ENABLE to 1

2E0LXY, 2020-04-04:
If I have a running version of XLX already installed on my VPS how do I set a default module for YSF please?

I understand normally I would  need to enable it in the src/main.h file, by setting the option YSF_AUTOLINK_ENABLE to 1 and adapt YSF_AUTOLINK_MODULE to my needs but the server is already running.

Many thanks

Daren

narspt, 2020-04-04:
You need to stop it, recompile and start again :)  main.h settings will be hardcoded on your compiled binary...

yo2loj, 2020-04-05:
At least give complete instructions, and not only half of the data.
- Recompile the binary after the needed changes (by running just a 'make clean' and 'make ', WITHOUT the 'make install' in the src directory)
- Stop your running xlxd (command 'service xlxd stop')
- Copy your freshly compiled 'xlxd' binary from the src folder over the old binary (located in /xlxd)
- Start the server, using 'service xlxd start'

2E0LXY, 2020-04-05:
Many thanks for all of your help.
IT WORKS!!!!
I could never have done this without your help, you are a genius! 
http://xlx899.mywire.org

closed 2020-07-27 by LX1IQ