LX3JL/xlxd #168: Audio lost and only fix is to restart

sm0tsc, 2020-05-07:
Hi since some time xlxd stops to pass audio. Callsigns etc is passed
Any clue how to debug? Or fix. I don´t  find anything specific in logs. XLX010 is using last version
//SM0TSC

LX1IQ, 2020-05-08:
Hi,
Normally it is not the XLX, which is crashing, but the usb port of the AMBed server.
You should see very high packet loses in the log.
Most of the time it is sufficient to kill the AMBed task and start it again.
In the download section, you can find a small script, which allows you to run the AMBed as a service.
You can schedule a restart of the service at night.
We need to add some sort of watchdog to the AMBed. So if the dongle get unresponsive the service can restarts by its own.

73, Luc

sm0tsc, 2020-05-08:
This also occurs for d-star to d-star connections
//Johan

LX1IQ, 2020-05-08:
Do to the fact that all traffic is always sent to the AMBed server. All modes are affected.

73, Luc

sm0tsc, 2020-05-08:
Ok can you give some example on how the logentry will look like?

//Johan

sm0tsc, 2020-05-08:
I ususally restart xlxd wich makes it work again.. ambeD is on another site
//Johan

LX1IQ, 2020-05-08:
You are going to see in the XLX log very high packet lost for the AMBed.
If you kill the AMbed, it takes a few seconds until the XLX disconnects the AMBed server with a keep alive timeout.
At this point, D-Star to D-Star should already come back.

73, Luc

sm0tsc, 2020-05-08:
Why does it get "fixed" by restarting xlxd then? I usually restart only
xlxd when this happens. As ambed is on another host in my lab

//Johan

LX1IQ, 2020-05-08:
If you stop the XLX, of course the connection on the AMBed side is disconnecting too.
When you start the XLX back on, a new connection is established to the AMBed server which is resetting the the dongle connection, connected to the AMbed as well.

sm0tsc, 2020-05-08:
Tnx info
I will do a script restarting ambed.. Hopefully we can monitor it somehow
in the future
//Johan

sm0tsc, 2020-05-08:
Seems that the easiest way will be adding this to the systemd script

Restart=always
RuntimeMaxSec=604800 (and a suitable number of seconds)

//SM0TSC

sm0tsc, 2020-05-08:
My ambed.service as an example

[Unit]
Description=Ambed Service
Requires=network.target
After=syslog.target network.target

[Service]
Type=simple
Restart=always
RuntimeMaxSec=14400
RestartSec=3
StandardOutput=null
WorkingDirectory=/usr/local/bin
ExecStartPre=-/sbin/rmmod ftdi_sio
ExecStartPre=-/sbin/rmmod usbserial
ExecStart=/usr/bin/screen -S ambed -D -m /usr/local/bin/ambed 192.168.2.203
ExecStop=/usr/bin/screen -S ambed -X quit

[Install]
WantedBy=multi-user.target

sm0tsc, 2020-05-10:
Seems not to help. Only helps with restarting xlxd.. Any more ideas?

//Johan

LX1IQ, 2020-05-10:
Any errors in the log ?

sm0tsc, 2020-05-10:
Nothing that I can see out of the ordinary I see some ambed timeouts but other errors. Restarted ambed with no change, then after some mnutes I restarted xlxd and trafic started flowing again

//Johan

bv3un, 2021-02-24:
same problem here .when the audio not pass through .the "timed out" is appear on /var/log/message 
after restart ambed .it becomes normal . xlxd is 2.4.1 /ambed  1.3.4 or 1.3.5 same problem

ShaYmez, 2022-01-07:
The trouble is, we can put a watchdog on the AMBEd service but the XLXd stream still does not recover. You have to physically restart the XLXD service to recover! Is there a fix for this yet? Maybe add a watchdog for XLX to restart in the event of a failure?