LX3JL/xlxd #178: no YSF-stream with PiStar-Hotspot which is connected to XLX

spitzbube80, 2020-11-27:
Hello Everbody,

I have a little issue with YSF on my XLXd.

My PiStar-Hotspot is correctly connected with my XLX in YSF on Module B. I can see my Hotspot and the connected module on my Dashboard.  I can hear everybody on this Modul, but nobody can hear me. In the logfile of my xlxd I get the Message: 

`xlxd: Deleting oprhaned Last Frame Packet with StreamId 2025 from nnn.nnn.nnn.nnn` 

I installed the latest github-version of xlxd with the Fix for "Deleting oprhaned Last Frame Packet" on my Server.
 
I have no issues with DMR & D-Star , only with YSF. 

Does anyone have an idea what my mistake is ?

PS: sorry for my bad englisch ;-) 
vy73

xe1dvi, 2020-11-27:
Make sure that the call sign on your Yaesu radio only was that, nothing else like your name. ONLY call sign.

Sincerely,
Rey.XE1DVI / W6DVI

yo2loj, 2020-11-27:
I think an explanation is needed on this:

The YSF part is handled internally as a DMR connection, thus the system 
needs to translate the callsign to a valid DMR id using an (downloaded) 
lookup table.
This means that your callsign set in the C4FM radio needs to DMR 
registered and be IDENTICAL to the one registered in DMR (so no /DAN 
/MIKE or /C4FM at the end). Otherwise it will be dropped.

Marius, YO2LOJ

spitzbube80, 2020-11-29:
many THXs to XE1DVI & YO2LOJ. You're so great :-)

The Callsign was my issue. Everything looks good now.

vy73 de Christian (DL9CMA)

closed 2020-11-29 by spitzbube80