LX3JL/xlxd #180: Move config options in main.h to a config file

marrold, 2021-01-16:
Hi,

Firstly, thanks for making XLX. I am working on Dockerising xlxd and ambed and both are working, however the dependency on configuring things in `main.h` at build time makes it more difficult to make a generic container that anyone can use.

Would it be possible to shift these parameters into a .ini file instead? Unfortunately my c++ is lacking.

Thanks
Matthew
2E0SIP

yo2loj, 2021-01-16:
Matthew, I think that's actually counterproductive.
Hosting a public reflector should require minimal computing and networking skill, otherwise it will lead to chaos in the current existing system affecting all users.
Changing/compiling is a kind of shield for those not being able to grasp and do that.
This is not a user level software, but an infrastructure one, so not everyone should be able to use it.

marrold, 2021-01-17:
I'm not really sure how to respond to that, other than I disagree with gatekeeping.

yo2loj, 2021-01-17:
That is only my personal opinion, not an official position of any kind. In the end it is Luc's decision.
But at the moment, portability is basically ensured by compiling the whole code against libc as the only dependency.
The problem I see is actually docker itself, which may complicate the whole network setup, but again, it is my personal opinion.
At the moment, the installation itself involves placing 4 configuration files and the compiled binary in a folder and firing it up, with the default parameters mostly doing the trick, and as a secondary step to install a dashboard, again just a copy operation into the web server location, with a few edits in a file.
Do you really think there is an added benefit of having this in a docker container? Is it really that portable, especially across OS and docker versions and different controllers, or do we shoot ourselves in the foot by committing to maintain a lot of precompiled HW dependent versions?
At the moment it compiles on basically any Linux system. What is the added value compared to the maintenance effort on the development side?

citynetcomputing, 2021-01-27:
> 
> 
> I'm not really sure how to respond to that, other than I disagree with gatekeeping.
> 
> Of course the maintainers are entitled to their opinions and the direction of the project, but my _main_ goal is to make the XLX installation portable and reproducible rather than to make it easier for people with little experience to run their own reflector. Thats a secondary benefit.

I agree with this. As an admin managing multiple installations it would be way easier to test and deploy without having to compile this every time. Very counter productive managing multiple sites.