LX3JL/xlxd #214: XLX to HBlink issues.

citynetcomputing, 2022-03-06:
When connecting XLX to HBlink as XLXPEER. If a user on the XLX side does NOT have a registered DMR id then XLX presents the DMR id of "0" to HBlink which breaks the DMR side. Need a way to specify/present alternate DMR ID to HBlink if the XLX user is not registered on DMR.

Thx
73

yo2loj, 2022-03-06:
One of the mechanism to ensure a valid DMR id to user mapping is to use 
an registered ID.

Isn't it correct to expect a registered ID of the DMR side?

The only workaround should be to use a valid, registered DMR id on the 
XLX side and to prevent DMR communications for unregistered callsigns.

Otherwise we will end to have a ban on XLX nodes in DMR networks for not 
ensuring valid user identification.

Just my 2c...

Marius, YO2LOJ

On 06/03/2022 20:21, Jay wrote:
> When connecting XLX to HBlink as XLXPEER. If a user on the XLX side 
> does NOT have a registered DMR id then XLX presents the DMR id of "0" 
> to HBlink which breaks the DMR side. Need a way to specify/present 
> alternate DMR ID to HBlink if the XLX user is not registered on DMR.

citynetcomputing, 2022-03-06:
There not on the DMR side. They are on the D-STAR side hence the problem. Not all D-STAR users have DMR or DMR id's. Brandmeister fixes this issue by converting the "0" talkgroup through there system to your Group Talkgroup ie: 310441 for us. So if a user on D-STAR dont have a DMR ID it will come through to the DMR side as 310441.

yo2loj, 2022-03-06:
Yes, this is exactly what I was talking about. A user on the D-star side 
without a DMR ID should NOT have access to DMR, since on the DMR network 
the user is uniquely, and i think legally, identified by his DMR id.

IMHO the same should apply to any D-Star/C4FM to DMR gateways (e.g. on 
my reflectors, this is enforced in the YSF2DMR), converting to a 
"generic" node ID being actually identity spoofing, circumventing user 
identification.